anholt/linux #92: vc4-kms-v3d gives black screen on tv/monitor (recognized as touchscreen?) | vc4-fkms-v3d works

pkoretic, 2017-03-06:
Hi all!

I've decided to test latest vc4 driver from 4.9 branch 935c7ce84c982a26f567a03a58a1537424569938

I've also tried the same on archarm linux that I have, kernel 4.9.13.

I've added to:
config.txt
```
dtoverlay=vc4-kms-v3d
gpu_mem=256
```
cmdline.txt
```cma-128```

after reboot I have a black screen. I've tested on LG TV and Samsung monitor (over HDMI).
changing to `dtoverlay=vc4-fkms-v3d` boots fine

difference that I can see is that in former case my display is detected as touch screen?
Is that expected in current state? 
I see some have got it working, and I've tried different configuration options like `hdmi_safe` and others to no avail.

should vc4-kms-v3d give better performance than vc4-fkms-v3d? In my case, I'm only running one app using OpenGL ES (without X) and using vc4-kms-v3d app runs noticeably slower than using broadcom driver so I'm curious if this is the issues of the f/kms and can be improved.

```
Jan  1 00:00:03 leti user.warn kernel: [    2.696588] ------------[ cut here ]------------
Jan  1 00:00:03 leti user.warn kernel: [    2.696730] WARNING: CPU: 0 PID: 130 at drivers/gpu/drm/drm_atomic_helper.c:1140 drm_atomic_helper_wait_for_vblanks+0x268/0x26c [drm_kms_helper]
Jan  1 00:00:03 leti user.warn kernel: [    2.696760] [CRTC:65] vblank wait timed out
Jan  1 00:00:03 leti user.warn kernel: [    2.696815] Modules linked in: brcmfmac panel_raspberrypi_touchscreen(+) vc4(+) brcmutil drm_kms_helper cfg80211 drm joydev rfkill syscopyarea sysfillrect sysimgblt fb_sys_fops i2c_gpio i2c_bcm283
Jan  1 00:00:03 leti user.warn kernel: [    2.696825] CPU: 0 PID: 130 Comm: udevd Not tainted 4.9.13-v7 #1
Jan  1 00:00:03 leti user.warn kernel: [    2.696828] Hardware name: BCM2835
Jan  1 00:00:03 leti user.warn kernel: [    2.696863] [<8010fb14>] (unwind_backtrace) from [<8010c018>] (show_stack+0x20/0x24)
Jan  1 00:00:03 leti user.warn kernel: [    2.696878] [<8010c018>] (show_stack) from [<80455be8>] (dump_stack+0xc8/0x10c)
Jan  1 00:00:03 leti user.warn kernel: [    2.696896] [<80455be8>] (dump_stack) from [<8011d1b8>] (__warn+0xf8/0x110)
Jan  1 00:00:03 leti user.warn kernel: [    2.696913] [<8011d1b8>] (__warn) from [<8011d218>] (warn_slowpath_fmt+0x48/0x50)
Jan  1 00:00:03 leti user.warn kernel: [    2.697032] [<8011d218>] (warn_slowpath_fmt) from [<7f267218>] (drm_atomic_helper_wait_for_vblanks+0x268/0x26c [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.697227] [<7f267218>] (drm_atomic_helper_wait_for_vblanks [drm_kms_helper]) from [<7f2af688>] (vc4_atomic_complete_commit+0x64/0x8c [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.697354] [<7f2af688>] (vc4_atomic_complete_commit [vc4]) from [<7f2af87c>] (vc4_atomic_commit+0x1cc/0x25c [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.697671] [<7f2af87c>] (vc4_atomic_commit [vc4]) from [<7f094644>] (drm_atomic_commit+0x5c/0x68 [drm])
Jan  1 00:00:03 leti user.warn kernel: [    2.698010] [<7f094644>] (drm_atomic_commit [drm]) from [<7f26d148>] (restore_fbdev_mode+0x160/0x2b8 [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.698223] [<7f26d148>] (restore_fbdev_mode [drm_kms_helper]) from [<7f26e228>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x40/0x84 [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.698438] [<7f26e228>] (drm_fb_helper_restore_fbdev_mode_unlocked [drm_kms_helper]) from [<7f26e2ac>] (drm_fb_helper_set_par+0x40/0x70 [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.698554] [<7f26e2ac>] (drm_fb_helper_set_par [drm_kms_helper]) from [<80492954>] (fbcon_init+0x57c/0x5c8)
Jan  1 00:00:03 leti user.warn kernel: [    2.698573] [<80492954>] (fbcon_init) from [<804d3efc>] (visual_init+0xd0/0x118)
Jan  1 00:00:03 leti user.warn kernel: [    2.698589] [<804d3efc>] (visual_init) from [<804d5520>] (do_bind_con_driver+0x1bc/0x39c)
Jan  1 00:00:03 leti user.warn kernel: [    2.698604] [<804d5520>] (do_bind_con_driver) from [<804d5aec>] (do_take_over_console+0x160/0x1cc)
Jan  1 00:00:03 leti user.warn kernel: [    2.698620] [<804d5aec>] (do_take_over_console) from [<80492a20>] (do_fbcon_takeover+0x80/0xdc)
Jan  1 00:00:03 leti user.warn kernel: [    2.698636] [<80492a20>] (do_fbcon_takeover) from [<804966d8>] (fbcon_event_notify+0x908/0x94c)
Jan  1 00:00:03 leti user.warn kernel: [    2.698654] [<804966d8>] (fbcon_event_notify) from [<8013e43c>] (notifier_call_chain+0x54/0x94)
Jan  1 00:00:03 leti user.warn kernel: [    2.698671] [<8013e43c>] (notifier_call_chain) from [<8013e874>] (__blocking_notifier_call_chain+0x58/0x70)
Jan  1 00:00:03 leti user.warn kernel: [    2.698688] [<8013e874>] (__blocking_notifier_call_chain) from [<8013e8b4>] (blocking_notifier_call_chain+0x28/0x30)
Jan  1 00:00:03 leti user.warn kernel: [    2.698703] [<8013e8b4>] (blocking_notifier_call_chain) from [<8049824c>] (fb_notifier_call_chain+0x28/0x2c)
Jan  1 00:00:03 leti user.warn kernel: [    2.698716] [<8049824c>] (fb_notifier_call_chain) from [<8049a2b8>] (register_framebuffer+0x1a4/0x2a4)
Jan  1 00:00:03 leti user.warn kernel: [    2.698828] [<8049a2b8>] (register_framebuffer) from [<7f26e57c>] (drm_fb_helper_initial_config+0x2a0/0x46c [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.699040] [<7f26e57c>] (drm_fb_helper_initial_config [drm_kms_helper]) from [<7f26eef0>] (drm_fbdev_cma_init_with_funcs+0x90/0x108 [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.699256] [<7f26eef0>] (drm_fbdev_cma_init_with_funcs [drm_kms_helper]) from [<7f26ef90>] (drm_fbdev_cma_init+0x28/0x30 [drm_kms_helper])
Jan  1 00:00:03 leti user.warn kernel: [    2.699426] [<7f26ef90>] (drm_fbdev_cma_init [drm_kms_helper]) from [<7f2af9dc>] (vc4_kms_load+0x8c/0xc4 [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.699549] [<7f2af9dc>] (vc4_kms_load [vc4]) from [<7f2a9800>] (vc4_drm_bind+0x100/0x140 [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.699620] [<7f2a9800>] (vc4_drm_bind [vc4]) from [<804f6138>] (try_to_bring_up_master+0x180/0x1bc)
Jan  1 00:00:03 leti user.warn kernel: [    2.699638] [<804f6138>] (try_to_bring_up_master) from [<804f6404>] (component_master_add_with_match+0x9c/0xd0)
Jan  1 00:00:03 leti user.warn kernel: [    2.699707] [<804f6404>] (component_master_add_with_match) from [<7f2a9904>] (vc4_platform_drm_probe+0xc4/0xd4 [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.699778] [<7f2a9904>] (vc4_platform_drm_probe [vc4]) from [<804fd03c>] (platform_drv_probe+0x60/0xc0)
Jan  1 00:00:03 leti user.warn kernel: [    2.699793] [<804fd03c>] (platform_drv_probe) from [<804fb6e0>] (driver_probe_device+0x214/0x2c0)
Jan  1 00:00:03 leti user.warn kernel: [    2.699806] [<804fb6e0>] (driver_probe_device) from [<804fb854>] (__driver_attach+0xc8/0xcc)
Jan  1 00:00:03 leti user.warn kernel: [    2.699819] [<804fb854>] (__driver_attach) from [<804f96c0>] (bus_for_each_dev+0x78/0xac)
Jan  1 00:00:03 leti user.warn kernel: [    2.699832] [<804f96c0>] (bus_for_each_dev) from [<804faee0>] (driver_attach+0x2c/0x30)
Jan  1 00:00:03 leti user.warn kernel: [    2.699844] [<804faee0>] (driver_attach) from [<804fa9e4>] (bus_add_driver+0x11c/0x228)
Jan  1 00:00:03 leti user.warn kernel: [    2.699857] [<804fa9e4>] (bus_add_driver) from [<804fbe1c>] (driver_register+0x88/0x104)
Jan  1 00:00:03 leti user.warn kernel: [    2.699871] [<804fbe1c>] (driver_register) from [<804fcf88>] (__platform_driver_register+0x50/0x58)
Jan  1 00:00:03 leti user.warn kernel: [    2.699939] [<804fcf88>] (__platform_driver_register) from [<7f2c9064>] (vc4_drm_register+0x64/0x78 [vc4])
Jan  1 00:00:03 leti user.warn kernel: [    2.700011] [<7f2c9064>] (vc4_drm_register [vc4]) from [<80101bfc>] (do_one_initcall+0x54/0x180)
Jan  1 00:00:03 leti user.warn kernel: [    2.700026] [<80101bfc>] (do_one_initcall) from [<8020b850>] (do_init_module+0x74/0x1e4)
Jan  1 00:00:03 leti user.warn kernel: [    2.700043] [<8020b850>] (do_init_module) from [<801a3df0>] (load_module+0x1cd4/0x2314)
Jan  1 00:00:03 leti user.warn kernel: [    2.700060] [<801a3df0>] (load_module) from [<801a4648>] (SyS_finit_module+0xb8/0xc8)
Jan  1 00:00:03 leti user.warn kernel: [    2.700075] [<801a4648>] (SyS_finit_module) from [<80108060>] (ret_fast_syscall+0x0/0x1c)
Jan  1 00:00:03 leti user.warn kernel: [    2.700151] ---[ end trace 1a15e98f6cafef76 ]---```

lategoodbye, 2017-03-08:
Please look at #66 and try the linked patch.

anholt, 2017-03-09:
Please attach the output of `dmesg`, `xrandr --verbose`, and `cat /debug/clk/clk_summary` (with debugfs mounted) from the system when the display is broken.  If you can't get X up, `dmesg` with `drm.debug=0x1e` on the kernel command line would do.

vc4-kms-v3d doesn't care about any of your `config.txt` display settings (except `avoid_warnings=1` which is required).  If any of `config.txt` *did* affect vc4-kms-v3d, that would actually be a bug.  Also note that gpu_mem= is totally wasted on vc4-kms-v3d, and just steals memory from Linux.

vc4-kms-v3d and vc4-fkms-v3d should be the same performance for 3D stuff, assuming you're synced to vblank. You do miss out on async pageflips in fkms, though.

pkoretic, 2017-03-09:
Thanks for the info!

I tried now with the latest kernel from the anholt branch which includes mentioned patch in 45a3a972f07bd39fa9aa96aaa98517d6400b6fb6 commit if I'm correct

hasn't changed unfortunately, here are attached logs  
[clk_summary](https://gist.githubusercontent.com/pkoretic/ccc6243f71274ee02b6d13be52d61516/raw/2396cd77e7633bfb6045d6c6376fafa5625e39e0/debugfs.log) 
[dmesg](https://gist.githubusercontent.com/pkoretic/f8a6ad786b3748b9ad1196181c714e1c/raw/632fc63a96f17339d68de9b532a4137e6d66de40/dmesg.log)

hopefully I haven't missed something, If it helps I can also setup X and provide logs

anholt, 2017-03-10:
Oh, HDMI is probing as disconnected.  I think you're just hitting the regression in 5d1da107bc619c1cc091ee5301f85cd89a6d92fb that we talked about in https://github.com/raspberrypi/linux/pull/1869

anholt, 2017-06-13:
Could you retest with current kernel and firmware, since the HPD bug is supposed to be fixed now?

pkoretic, 2017-06-13:
Let me test this ASAP!

pkoretic, 2017-06-17:
@anholt works as expected using latest 4.11 (6a919d906e49fd2cfc2b7d813818cf493d1c4315)

closed 2017-06-17 by pkoretic