anholt/linux #93: Display goes out of Range mid way through running

tl8roy, 2017-03-09:
We have a Raspberry Pi 2 running Rasbian Jessie with the OpenGL driver.

It is connected to a HDMI to DisplayPort Adaptor and an odd size touch screen display (1280x800).

We are running lightdm and it starts up a full screen application.

This application is built using Rust and makes use of the OpenGL driver.

After this application has been running for about an hour, the display suddenly goes ```Out of Range```. After about 10 - 15 minutes, the screen then works again with no issues.

This is happening on 2 independent sets of hardware, although the SD cards are clones.

I have attached the output of xrandr.
[xrandr_output.txt](https://github.com/anholt/linux/files/829402/xrandr_output.txt)

dmesg has no message before or after the situation occurs.

This is the output from /debug/clk/clk_summary:
[clk_summary.txt](https://github.com/anholt/linux/files/829452/clk_summary.txt)

This is the same issue as mentioned [here](http://raspberrypi.stackexchange.com/questions/63011/display-goes-out-of-range).

lategoodbye, 2017-03-09:
@tl8roy Could you please also add the kernel version?

anholt, 2017-03-09:
In good state we have:

          pllh_pix                        1            1    70755000          0 0

and in bad state we have:

          pllh_pix                        1            1    53332500          0 0

Somebody has set PLLH on us, away from the 71.110MHz that your chosen mode requested.  The next step would be to add a debug message when we're setting PLLH to see if it's something inside Linux doing it, or if the firmware is breaking things.

[0001-clk-debug-print-when-PLL-rates-get-set.patch.txt](https://github.com/anholt/linux/files/832044/0001-clk-debug-print-when-PLL-rates-get-set.patch.txt)

Thinking of the firmware: You have `avoid_warnings=1` in your `config.txt`, right?  The firmware will break your display during undervoltage events without that.

Also, please attach dmesg -- even if you didn't see anything interesting in it, it's important for me to understand your situation.

Are you making any calls to the firmware that you know of?  Anything with MMAL, VCHI, vcdbg, tvservice, etc?

tl8roy, 2017-03-09:
Kernel Version is ```4.4.50-v7+```

dmesg:
[dmesg_output.txt](https://github.com/anholt/linux/files/832465/dmesg_output.txt)

I didn't have ```avoid_warnings=1``` set, I have changed that now.

I don't explicitly make any firmware calls.

Do you want me to build a kernel with that patch and run it?

anholt, 2017-03-10:
Yeah, that will let us know if it something in Linux at fault.

tl8roy, 2017-03-10:
I had a quick go at compiling before the weekend.

While the build worked, the display doesn't have any output (not even console)

I'll try to troubleshoot on Monday.

tl8roy, 2017-03-14:
I am having a bit of trouble with the kernel. It compiles ok, but when I try to run it on the pi, the screen is blank.

Is this a known issue in the current master?

tl8roy, 2017-03-14:
If I use the legacy driver, everything works as per normal.

anholt, 2017-03-14:
Same kernel version as the binary you were using before?

I don't know of any breakage on Pi2 -- Pi3 is currently broken in the downstream 4.9 tree because of the gpio-exp driver being wired up incorrectly.

tl8roy, 2017-03-14:
I was using 4.9, but I'll drop back to 4.4 and try that.

tl8roy, 2017-03-14:
Will the output appear in dmesg?

I managed to turn on debugging so dmesg now looks like this:
[dmesg.txt](https://github.com/anholt/linux/files/840246/dmesg.txt)

I don't know the exact time it went out of range, but there is something unusual around 03:46:00 (Line 1137)

anholt, 2017-03-14:
The thing at 3:46 is either something in your userspace asking to turn off the monitor, or possibly your userspace exiting and DRM trying to restore the fbdev state (which is presumably now screen off due to idle).

tl8roy, 2017-03-14:
Is there a way to work out what process is triggering this?
The pi display stack is pretty simple, Xorg, LightDM and then the program that is running. X is set to not sleep or turn the display off.

anholt, 2017-03-15:
Breakpoint in X on `drmmode_output_dpms()` might find a caller

tl8roy, 2017-03-15:
Unfortunately, I think that is above my skill level.

anholt, 2017-03-15:
if you can ssh in while X is running, it should hopefully be a matter of

    # sudo gdb /usr/bin/Xorg `pidof Xorg`
    (gdb) b drmmode_output_dpms
    (gdb) c

and then when that triggers:

    (gdb) bt
    (gdb) c

If after 'c' the screen goes black, then 'bt' will have given us some information on who turned off the screen (X's DPMS code, a client configuring the screens, etc.).

tl8roy, 2017-03-22:
Trying this today. So fair nothing has happened for a few hours. I am using the kernel that I built.

tl8roy, 2017-03-23:
X is definitely not triggering the issue with that function.

tl8roy, 2017-03-26:
Some good news. I upgraded the Kernel to 4.9 via ```rpi-update```. I have run it over the weekend and it appears that the issue has gone away. I want to test it for a few more hours, but it looks like that one of the bug fixes done between 4.4 and 4.9 has resolved the issue.

closed 2017-03-27 by tl8roy