anholt/linux #94: cma_alloc issue mainline 4.11 rc2

ric96, 2017-03-18:
```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.11.0-rc2 (pi@raspberrypi) (gcc version 4.9.2 (Raspbian 4.9.2-10) ) #1 SMP Sat Mar 18 14:39:21 UTC 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 64 MiB at 0x2b000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 192512
[    0.000000] free_area_init_node: node 0, pgdat c1527e80, node_mem_map ea9f9000
[    0.000000]   DMA zone: 1504 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 192512 pages, LIFO batch:31
[    0.000000] percpu: Embedded 16 pages/cpu @ea9ad000 s34432 r8192 d22912 u65536
[    0.000000] pcpu-alloc: s34432 r8192 d22912 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 191008
[    0.000000] Kernel command line: earlyprintk console=serial0,115200 console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 678088K/770048K available (10240K kernel code, 1189K rwdata, 4012K rodata, 2048K init, 536K bss, 26424K reserved, 65536K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xef800000 - 0xff800000   ( 256 MB)
    lowmem  : 0xc0000000 - 0xef000000   ( 752 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0208000 - 0xc0d00000   (11232 kB)
      .init : 0xc1200000 - 0xc1400000   (2048 kB)
      .data : 0xc1400000 - 0xc15297a0   (1190 kB)
       .bss : 0xc152b000 - 0xc15b1278   ( 537 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU debugfs-based tracing is enabled.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=16 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=4
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000030] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000069] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000166] bcm2835: system timer (irq = 33)
[    0.001854] arm_arch_timer: WARNING: Invalid trigger for IRQ16, assuming level low
[    0.001865] arm_arch_timer: WARNING: Please fix your firmware
[    0.001879] arm_arch_timer: WARNING: Invalid trigger for IRQ17, assuming level low
[    0.001888] arm_arch_timer: WARNING: Please fix your firmware
[    0.002040] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.002064] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.002088] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.002109] Switching to timer-based delay loop, resolution 52ns
[    0.002561] Console: colour dummy device 80x30
[    0.003588] console [tty0] enabled
[    0.003657] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.003721] pid_max: default: 32768 minimum: 301
[    0.003936] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003977] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.004952] CPU: Testing write buffer coherency: ok
[    0.005612] CPU0: update cpu_capacity 1024
[    0.005658] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.006066] Setting up static identity map for 0x300000 - 0x300098
[    0.008379] EFI services will not be available.
[    0.008863] smp: Bringing up secondary CPUs ...
[    0.009885] arm_arch_timer: WARNING: Invalid trigger for IRQ16, assuming level low
[    0.009889] arm_arch_timer: WARNING: Please fix your firmware
[    0.009900] arm_arch_timer: WARNING: Invalid trigger for IRQ17, assuming level low
[    0.009903] arm_arch_timer: WARNING: Please fix your firmware
[... 371 lines cut ...]
[   92.285895] [drm:vc4_bo_create] *ERROR* Failed to allocate from CMA:
[   92.285900] [drm] num bos allocated: 38
[   92.285906] [drm] size bos allocated: 61436kb
[   92.285911] [drm] num bos used: 37
[   92.285915] [drm] size bos used: 60372kb
[   92.285919] [drm] num bos cached: 1
[   92.285924] [drm] size bos cached: 1064kb
[   92.285942] [drm:vc4_validate_bin_cl] *ERROR* 0x00000000: packet 112 (VC4_PACKET_TILE_BINNING_MODE_CONFIG) failed to validate
[   93.013040] cma: cma_alloc: alloc failed, req-size: 1024 pages, ret: -12
[   93.013066] vc4-drm soc:gpu: failed to allocate buffer with size 4194304
[   93.013748] cma: cma_alloc: alloc failed, req-size: 1024 pages, ret: -12
[   93.013761] vc4-drm soc:gpu: failed to allocate buffer with size 4194304
[   93.013784] [drm:vc4_bo_create] *ERROR* Failed to allocate from CMA:
[   93.013791] [drm] num bos allocated: 54
[   93.013796] [drm] size bos allocated: 62520kb
[   93.013800] [drm] num bos used: 53
[   93.013805] [drm] size bos used: 58424kb
[   93.013809] [drm] num bos cached: 1
[   93.013813] [drm] size bos cached: 4096kb
[   93.014517] cma: cma_alloc: alloc failed, req-size: 1024 pages, ret: -12
[   93.014531] vc4-drm soc:gpu: failed to allocate buffer with size 4194304
=====================
Linux raspberrypi 4.11.0-rc2 #1 SMP Sat Mar 18 14:39:21 UTC 2017 armv7l GNU/Linux
```
Also does the stable 4.10 branch work properly?

anholt, 2017-03-20:
[    0.000000] cma: Reserved 64 MiB at 0x2b000000

You're going to need more than 64MB of CMA to do graphics.  I would recommend using Raspbian or another distribution configured to actually support graphics on Raspberry Pi, rather than trying to roll your own.

closed 2017-03-20 by anholt

ric96, 2017-03-20:
I am using raspbian, and i did have gpu_mem=256
but i was booting using uboot
```
mmc dev 0
setenv fdtfile bcm2836-rpi-b-2.dtb
setenv bootargs earlyprintk console=serial0,115200 console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootwait
fatload mmc 0:1 ${kernel_addr_r} zImage
fatload mmc 0:1 ${fdt_addr_r} ${fdtfile}
bootz ${kernel_addr_r} - ${fdt_addr_r}
```

config.txt
```
gpu_mem=256
dtoverlay=vc4-kms-v3d
```

anholt, 2017-03-20:
gpu_mem is not cma, and just steals memory from Linux when vc4 is enabled.

ric96, 2017-03-21:
cma=256M ?

anholt, 2017-03-21:
That will misalign the area and probably cause system stability problems.  You should replicate what Raspbian does.

ric96, 2017-03-23:
how does raspbian do it? I did try it only on raspbian without uboot. Is the mkimage script also necessary?