anholt/linux #95: vc4: NULL pointer dereference in 4.11.0-rc3-next-20170324+

lategoodbye, 2017-03-25:
Hi Eric,
i'm getting a NULL pointer dereference in current linux-next ( 4.11.0-rc3-next-20170324+ ) on my Raspberry Pi Zero during boot. This issue is only reproducible in case HDMI is not connected. In order to see the oops FRAMEBUFFER_CONSOLE must be disabled.

```
[    2.516720] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[    2.524960] pgd = c0004000
[    2.527726] [00000004] *pgd=da6c000000000000
[    2.532070] Internal error: Oops: 805 [#1] ARM
[    2.536585] CPU: 0 PID: 49 Comm: kworker/0:2 Not tainted 4.11.0-rc3-next-20170324+ #7
[    2.544510] Hardware name: BCM2835
[    2.547995] Workqueue: events output_poll_execute
[    2.552778] task: da639900 task.stack: da6c0000
[    2.557374] PC is at rm_hole+0x20/0x68
[    2.561186] LR is at drm_mm_remove_node+0x34/0x2f4
[    2.566040] pc : [<c042c83c>]    lr : [<c042d05c>]    psr: 20000093
[    2.566040] sp : da6c1d40  ip : da6c1d58  fp : da6c1d54
[    2.577606] r10: 00000000  r9 : da6c7a74  r8 : ffffffe4
[    2.582901] r7 : 00000000  r6 : da6d8dc8  r5 : da6d2010  r4 : da6d8dc8
[    2.589482] r3 : 00000000  r2 : 00000000  r1 : 55555555  r0 : da6d8dc8
[    2.596065] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[    2.603368] Control: 00c5387d  Table: 1a784008  DAC: 00000051
[    2.609167] Process kworker/0:2 (pid: 49, stack limit = 0xda6c0208)
[    2.615489] Stack: (0xda6c1d40 to 0xda6c2000)
[    2.619912] 1d40: 55555555 da6d8c00 da6c1d84 da6c1d58 c042d05c c042c828 00000055 da6d8c00
[    2.628183] 1d60: da6d2010 da6b3410 20000013 da6c7800 da6c7a74 0000001c da6c1da4 da6c1d88
[    2.636455] 1d80: c0446ad4 c042d034 da769840 00000038 00000002 00000000 da6c1dd4 da6c1da8
[    2.644731] 1da0: c0438550 c0446a98 00000000 da769840 da6c7800 da6a23c0 da6c7800 3fffffff
[    2.653003] 1dc0: da6c7a74 00000003 da6c1de4 da6c1dd8 c043866c c0438490 da6c1dfc da6c1de8
[    2.661274] 1de0: c0438690 c0438644 00000000 da769840 da6c1e44 da6c1e00 c041ecb8 c043867c
[    2.669545] 1e00: c0437c80 c0437618 00000000 c0437d4c 00000028 da6c7a78 da6c7800 da6a23c0
[    2.677817] 1e20: da6c7988 da6c7ab0 da6c7800 da6c7800 da6c1ee8 da6c7988 da6c1e64 da6c1e48
[    2.686085] 1e40: c042157c c041eb70 00000000 da6c7988 da6c7ab0 00000001 da6c1e7c da6c1e68
[    2.694357] 1e60: c042160c c042154c dafc0d20 da6a23c0 da6c1e94 da6c1e80 c042152c c04215c8
[    2.702631] 1e80: da6c7800 00000001 da6c1ea4 da6c1e98 c0422060 c042149c da6c1eb4 da6c1ea8
[    2.710901] 1ea0: c044c68c c042204c da6c1ecc da6c1eb8 c0414288 c044c678 00000000 00000000
[    2.719174] 1ec0: da6c1f1c da6c1ed0 c04143c0 c0414260 00000000 c0149744 da6c1efc da796000
[    2.727444] 1ee0: 00000000 00000002 00000000 00000000 da6c1f44 da69c600 da6c7ab0 c0c0e058
[    2.735715] 1f00: c0c16dc0 dac60100 00000000 00000000 da6c1f54 da6c1f20 c01362d8 c0414298
[    2.743984] 1f20: c0c0e058 c0c0e058 c0c16dc0 da69c600 c0c0e058 c0c0e07c c0c16dc0 da69c618
[    2.752254] 1f40: da69cd28 00000008 da6c1f7c da6c1f58 c01367c8 c01360f8 da69cd00 da69e280
[    2.760525] 1f60: 00000000 da4abea4 da69c600 c0136524 da6c1fac da6c1f80 c013c068 c0136530
[    2.768793] 1f80: da6c0000 da69e280 c013bf20 00000000 00000000 00000000 00000000 00000000
[    2.777060] 1fa0: 00000000 da6c1fb0 c0108398 c013bf2c 00000000 00000000 00000000 00000000
[    2.785331] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    2.793600] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555555 55555555
[    2.801903] [<c042c83c>] (rm_hole) from [<c042d05c>] (drm_mm_remove_node+0x34/0x2f4)
[    2.809775] [<c042d05c>] (drm_mm_remove_node) from [<c0446ad4>] (vc4_crtc_destroy_state+0x48/0x68)
[    2.818873] [<c0446ad4>] (vc4_crtc_destroy_state) from [<c0438550>] (drm_atomic_state_default_clear+0xcc/0x1b4)
[    2.829086] [<c0438550>] (drm_atomic_state_default_clear) from [<c043866c>] (drm_atomic_state_clear+0x34/0x38)
[    2.839197] [<c043866c>] (drm_atomic_state_clear) from [<c0438690>] (__drm_atomic_state_free+0x20/0x68)
[    2.848712] [<c0438690>] (__drm_atomic_state_free) from [<c041ecb8>] (restore_fbdev_mode+0x154/0x264)
[    2.858056] [<c041ecb8>] (restore_fbdev_mode) from [<c042157c>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x3c/0x7c)
[    2.868792] [<c042157c>] (drm_fb_helper_restore_fbdev_mode_unlocked) from [<c042160c>] (drm_fb_helper_set_par+0x50/0x6c)
[    2.879780] [<c042160c>] (drm_fb_helper_set_par) from [<c042152c>] (drm_fb_helper_hotplug_event+0x9c/0xb0)
[    2.889554] [<c042152c>] (drm_fb_helper_hotplug_event) from [<c0422060>] (drm_fbdev_cma_hotplug_event+0x20/0x24)
[    2.899858] [<c0422060>] (drm_fbdev_cma_hotplug_event) from [<c044c68c>] (vc4_output_poll_changed+0x20/0x24)
[    2.909828] [<c044c68c>] (vc4_output_poll_changed) from [<c0414288>] (drm_kms_helper_hotplug_event+0x34/0x38)
[    2.919880] [<c0414288>] (drm_kms_helper_hotplug_event) from [<c04143c0>] (output_poll_execute+0x134/0x16c)
[    2.929746] [<c04143c0>] (output_poll_execute) from [<c01362d8>] (process_one_work+0x1ec/0x3f8)
[    2.938564] [<c01362d8>] (process_one_work) from [<c01367c8>] (worker_thread+0x2a4/0x438)
[    2.946861] [<c01367c8>] (worker_thread) from [<c013c068>] (kthread+0x148/0x168)
[    2.954385] [<c013c068>] (kthread) from [<c0108398>] (ret_from_fork+0x14/0x3c)
[    2.961716] Code: e8bd4000 e5902024 e5903028 e1a04000 (e5823004)
[    2.967875] ---[ end trace 219bbd01c0c887cc ]---
```

addr2line to the pc says:

linux-next/./include/linux/list.h:104

anholt, 2017-03-27:
Untested patch
[0001-drm-vc4-Allocate-the-right-amount-of-space-for-boot-.patch.txt](https://github.com/anholt/linux/files/873902/0001-drm-vc4-Allocate-the-right-amount-of-space-for-boot-.patch.txt)

lategoodbye, 2017-03-28:
Perfect, this fixed the issue.

anholt, 2017-03-28:
Sent for review.  Thanks for testing!

closed 2017-03-31 by lategoodbye