anholt/linux #96: able to crash Raspberry Pi X server

woodlloydwood, 2017-04-18:
Occasional Raspberry Pi model B user here. Just upgraded to PIXEL 1.2.

Discovered that the packages I maintain when run together cause the Pi X server to freeze. (savi and geomview, just:
sudo apt-get install geomview
sudo apt-get install savi
geomview -run savi
)

This appears to be a gradual decrease in functionality over time with recent Pi distro upgrades; geomview used to be able to render texturemapped spheres just fine, then it could only render half of a texturemapped Earth sphere but still work, now with PIXEL geomview attempting to draw just freezes the X server for all applications and locks everything up. Which appears to be a bad thing.

_mgopengl_ctxset: undefined option: 1596824

is the error; I don't see this behaviour on other platforms (x86_64 Linux, mostly, some Mac.) Where to report this to whoever has an interest in maintaining X on the Pi?

http://savi.sf.net/
http://www.geomview.org/

thanks.

anholt, 2017-07-28:
Have you turned on vc4 in config.txt?  Either `dtoverlay=vc4-kms-v3d` (native KMS) or `dtoverlay=vc4-fkms-v3d` (closed source display stack).  I don't have time to help you if you're not running the graphics driver I'm writing.

anholt, 2017-07-28:
(FWIW, it displays fine for me with vc4 enabled)