anholt/linux #98: Official RPi2 Touchscreen does not shows when using "vc4-kms-v3d" but HDMI works.

TheAifam5, 2017-05-24:
https://github.com/raspberrypi/firmware/issues/812
https://github.com/WebReflection/benja/issues/13

lategoodbye, 2017-05-24:
Can you please provide an exact kernel version or better the dmesg after boot?

TheAifam5, 2017-05-24:
The dmesg logs i can't provide right now. In the next week i will be able to send it. I can say that connected RPi2 screen is seen and binded to Compositor-1 with valid resolution.

Packages:
`linux-firmware-20161222`
`raspberrypi-firmware-20170127`
`linux-raspberrypi-4.4.45`
`raspberrypi-bootloader-20170131`
`raspberrypi-bootloader-x-20170131`

If you need i can upload them.

lategoodbye, 2017-05-25:
Can you please update to rpi-4.9?

anholt, 2017-06-20:
You'll need to attach the dmesg for us to do anything.  And, yes, use 4.9.

WebReflection, 2017-07-18:
FYI `vc4-fkms-v3d` fixed the issue out of the box.

The latest `vc4-kms-v3d` on 4.9 never showed up a thing on the display, only HDMI