anholt/linux #99: 7'' Raspberry display resolution automatically sets to 720x480 and screen have artifacts

soswow, 2017-05-25:
Please redirect me if this is a bad place.
Raspberry Pi 3. + Official screen 7''. Box says it should be 800x480, but it defaults to 720x480, which I assume causes these funky colours.
I've downloaded latest raspbian, did all the required things like apt-get update, upgrade etc.

[image]
[image]
[image]

```
$ uname -a
Linux raspberrypi3 4.9.29-v7+ #1000 SMP Sun May 21 20:11:31 BST 2017 armv7l GNU/Linux
```

```
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.29-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1000 SMP Sun May 21 20:11:31 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 194048
[    0.000000] free_area_init_node: node 0, pgdat 80c6ee00, node_mem_map 9e552000
[    0.000000]   Normal zone: 1706 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 194048 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @af533000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 192342
[    0.000000] Kernel command line: 8250.nr_uarts=0 cma=256M bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9577b1c2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 493040K/776192K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 21008K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xaf800000 - 0xff800000   (1280 MB)
    lowmem  : 0x80000000 - 0xaf600000   ( 758 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78fd4   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000335] Console: colour dummy device 80x30
[    0.001241] console [tty1] enabled
[    0.001289] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001360] pid_max: default: 32768 minimum: 301
[    0.001723] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001766] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002781] Disabling cpuset control group subsystem
[    0.002969] CPU: Testing write buffer coherency: ok
[    0.003039] ftrace: allocating 22386 entries in 66 pages
[    0.055540] CPU0: update cpu_capacity 1024
[    0.055605] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.055706] Setting up static identity map for 0x100000 - 0x100034
[    0.057942] CPU1: update cpu_capacity 1024
[    0.057950] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.058674] CPU2: update cpu_capacity 1024
[... 267 lines cut ...]
#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
start_x=1
gpu_mem=250
dtoverlay=vc4-kms-v3d

# Official Rapsberry Pi display touchscreen
dtoverlay=rpi-ft5406

# Raspberry Pi official display backlight driver
dtoverlay=rpi-backlight
```

soswow, 2017-05-25:
Someone mentioned something somewhere, that changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d`can help. I tried. It did help with pixelation. Now it is not there. But a resolution is still 720x480.

anholt, 2017-06-01:
I've seen similar artifacts, and have done a bunch of debug but not come up with anything yet.