anholt/linux #101: vc4: possible memleak in vc4_hdmi_connector_get_modes

lategoodbye, 2017-07-22:
Hi,
i started my Raspberry Pi Zero W (4.13-rc1) without HDMI connected and after booting i connected HDMI. After some delay the kmemleak reported the following:

```
unreferenced object 0xd81cea00 (size 256):
  comm "kworker/0:1", pid 11, jiffies 4294951235 (age 2991.460s)
  hex dump (first 32 bytes):
    00 ff ff ff ff ff ff 00 22 f0 56 29 01 01 01 01  ........".V)....
    22 16 01 03 80 34 20 78 2a fc 81 a4 55 4d 9d 25  "....4 x*...UM.%
  backtrace:
    [<c07f741c>] kmemleak_alloc+0x60/0xc8
    [<c024fa60>] __kmalloc_track_caller+0x184/0x2f4
    [<c0223bf8>] krealloc+0x64/0x9c
    [<c0480410>] drm_do_get_edid+0xec/0x234
    [<c04805a4>] drm_get_edid+0x4c/0x304
    [<c049f230>] vc4_hdmi_connector_get_modes+0x30/0x90
    [<c0460fe8>] drm_helper_probe_single_connector_modes+0x250/0x5b4
    [<c046e03c>] drm_setup_crtcs+0x9c/0xafc
    [<c046eedc>] drm_fb_helper_hotplug_event+0x90/0xb4
    [<c046fa10>] drm_fbdev_cma_hotplug_event+0x20/0x24
    [<c049bd24>] vc4_output_poll_changed+0x20/0x24
    [<c0460850>] drm_kms_helper_hotplug_event+0x34/0x38
    [<c0460cac>] output_poll_execute+0x124/0x15c
    [<c013af68>] process_one_work+0x328/0x674
    [<c013c034>] worker_thread+0x2d0/0x444
    [<c0142078>] kthread+0x14c/0x16c
```

lategoodbye, 2017-08-07:
It seems that a similiar leak is there in linux-next-20170807 (RPI3 32bit):

```
unreferenced object 0xed5d6200 (size 256):
  comm "systemd-udevd", pid 133, jiffies 4294938237 (age 1097.920s)
  hex dump (first 32 bytes):
    00 ff ff ff ff ff ff 00 22 f0 56 29 01 01 01 01  ........".V)....
    22 16 01 03 80 34 20 78 2a fc 81 a4 55 4d 9d 25  "....4 x*...UM.%
  backtrace:
    [<c08250a8>] drm_do_get_edid+0xf8/0x230
    [<c0826998>] drm_get_edid+0x34/0x2f4
    [<bf27bfa4>] vc4_hdmi_connector_get_modes+0x20/0x84 [vc4]
    [<c080850c>] drm_helper_probe_single_connector_modes+0x198/0x680
    [<c0815948>] drm_setup_crtcs+0x16c/0xacc
    [<c08162c8>] __drm_fb_helper_initial_config_and_unlock+0x20/0x3f0
    [<c0817370>] drm_fbdev_cma_init_with_funcs+0x78/0xfc
    [<bf2793d4>] vc4_kms_load+0x8c/0xb8 [vc4]
    [<bf274998>] vc4_drm_bind+0xe0/0x130 [vc4]
    [<c08840f8>] try_to_bring_up_master+0x13c/0x184
    [<c08842fc>] component_master_add_with_match+0x80/0xb8
    [<bf274a78>] vc4_platform_drm_probe+0x90/0xa8 [vc4]
    [<c088af54>] platform_drv_probe+0x4c/0xac
    [<c08892ac>] driver_probe_device+0x224/0x2fc
    [<c0889430>] __driver_attach+0xac/0xb0
    [<c0887738>] bus_for_each_dev+0x64/0x94
unreferenced object 0xed49b300 (size 256):
  comm "kworker/1:2", pid 59, jiffies 4294938369 (age 1096.610s)
  hex dump (first 32 bytes):
    00 ff ff ff ff ff ff 00 22 f0 56 29 01 01 01 01  ........".V)....
    22 16 01 03 80 34 20 78 2a fc 81 a4 55 4d 9d 25  "....4 x*...UM.%
  backtrace:
    [<c08250a8>] drm_do_get_edid+0xf8/0x230
    [<c0826998>] drm_get_edid+0x34/0x2f4
    [<bf27bfa4>] vc4_hdmi_connector_get_modes+0x20/0x84 [vc4]
    [<c080850c>] drm_helper_probe_single_connector_modes+0x198/0x680
    [<c0815948>] drm_setup_crtcs+0x16c/0xacc
    [<c0816d2c>] drm_fb_helper_hotplug_event.part.7+0x78/0xe8
    [<c0808b54>] output_poll_execute+0x68/0x16c
    [<c035d884>] process_one_work+0x154/0x47c
    [<c035dbe4>] worker_thread+0x38/0x554
    [<c03632b0>] kthread+0x12c/0x16c
    [<c0307cc0>] ret_from_fork+0x14/0x34
    [<ffffffff>] 0xffffffff

```

anholt, 2017-08-07:
Yeah, need to kfree(edid)

anholt, 2017-08-15:
Fixed in drm-misc-next

closed 2017-08-15 by anholt