anholt/linux #102: vc4: warning in vc4_crtc_update_dlist after connecting HDMI

lategoodbye, 2017-07-22:
Hi,
i started my Raspberry Pi Zero W (4.13-rc1) without HDMI connected and after booting i connected HDMI and get the following warning (without "drm/vc4: Fix VBLANK handling in crtc->enable() path" applied):

```
[  139.445831] ------------[ cut here ]------------
[  139.446146] WARNING: CPU: 0 PID: 3 at drivers/gpu/drm/vc4/vc4_crtc.c:628 vc4_crtc_atomic_flush+0x124/0x1b4
[  139.446175] Modules linked in: brcmfmac brcmutil bcm2835_rng rng_core
[  139.446309] CPU: 0 PID: 3 Comm: kworker/0:0 Not tainted 4.13.0-rc1+ #18
[  139.446331] Hardware name: BCM2835
[  139.446376] Workqueue: events output_poll_execute
[  139.446456] [<c011062c>] (unwind_backtrace) from [<c010d2dc>] (show_stack+0x20/0x24)
[  139.446501] [<c010d2dc>] (show_stack) from [<c07e4498>] (dump_stack+0x20/0x28)
[  139.446551] [<c07e4498>] (dump_stack) from [<c0120528>] (__warn+0xe4/0x110)
[  139.446594] [<c0120528>] (__warn) from [<c0120624>] (warn_slowpath_null+0x30/0x38)
[  139.446639] [<c0120624>] (warn_slowpath_null) from [<c0495d98>] (vc4_crtc_atomic_flush+0x124/0x1b4)
[  139.446700] [<c0495d98>] (vc4_crtc_atomic_flush) from [<c046a530>] (drm_atomic_helper_commit_planes+0x254/0x27c)
[  139.446752] [<c046a530>] (drm_atomic_helper_commit_planes) from [<c049ba64>] (vc4_atomic_complete_commit+0x58/0xd4)
[  139.446800] [<c049ba64>] (vc4_atomic_complete_commit) from [<c049bc7c>] (vc4_atomic_commit+0x19c/0x1bc)
[  139.446862] [<c049bc7c>] (vc4_atomic_commit) from [<c0485278>] (drm_atomic_commit+0x54/0x68)
[  139.446909] [<c0485278>] (drm_atomic_commit) from [<c046c890>] (restore_fbdev_mode+0x110/0x268)
[  139.446956] [<c046c890>] (restore_fbdev_mode) from [<c046ef3c>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x3c/0x80)
[  139.447001] [<c046ef3c>] (drm_fb_helper_restore_fbdev_mode_unlocked) from [<c046efd0>] (drm_fb_helper_set_par+0x50/0x6c)
[  139.447042] [<c046efd0>] (drm_fb_helper_set_par) from [<c046eeec>] (drm_fb_helper_hotplug_event+0xa0/0xb4)
[  139.447085] [<c046eeec>] (drm_fb_helper_hotplug_event) from [<c046fa10>] (drm_fbdev_cma_hotplug_event+0x20/0x24)
[  139.447128] [<c046fa10>] (drm_fbdev_cma_hotplug_event) from [<c049bcdc>] (vc4_output_poll_changed+0x20/0x24)
[  139.447173] [<c049bcdc>] (vc4_output_poll_changed) from [<c0460850>] (drm_kms_helper_hotplug_event+0x34/0x38)
[  139.447217] [<c0460850>] (drm_kms_helper_hotplug_event) from [<c0460cac>] (output_poll_execute+0x124/0x15c)
[  139.447265] [<c0460cac>] (output_poll_execute) from [<c013af68>] (process_one_work+0x328/0x674)
[  139.447309] [<c013af68>] (process_one_work) from [<c013c034>] (worker_thread+0x2d0/0x444)
[  139.447359] [<c013c034>] (worker_thread) from [<c0142078>] (kthread+0x14c/0x16c)
[  139.447412] [<c0142078>] (kthread) from [<c0108330>] (ret_from_fork+0x14/0x24)
[  139.447444] ---[ end trace 899beaad1f7c68a5 ]---
```

After applying "drm/vc4: Fix VBLANK handling in crtc->enable() path" i get a different warning:

```
[  139.445847] ------------[ cut here ]------------
[  139.446168] WARNING: CPU: 0 PID: 11 at drivers/gpu/drm/vc4/vc4_crtc.c:535 vc4_crtc_update_dlist+0x50/0xd8
[  139.446198] Modules linked in: brcmfmac brcmutil bcm2835_rng rng_core
[  139.446332] CPU: 0 PID: 11 Comm: kworker/0:1 Not tainted 4.13.0-rc1+ #17
[  139.446355] Hardware name: BCM2835
[  139.446399] Workqueue: events output_poll_execute
[  139.446477] [<c011062c>] (unwind_backtrace) from [<c010d2dc>] (show_stack+0x20/0x24)
[  139.446522] [<c010d2dc>] (show_stack) from [<c07e44d8>] (dump_stack+0x20/0x28)
[  139.446570] [<c07e44d8>] (dump_stack) from [<c0120528>] (__warn+0xe4/0x110)
[  139.446613] [<c0120528>] (__warn) from [<c0120624>] (warn_slowpath_null+0x30/0x38)
[  139.446657] [<c0120624>] (warn_slowpath_null) from [<c0495cc4>] (vc4_crtc_update_dlist+0x50/0xd8)
[  139.446700] [<c0495cc4>] (vc4_crtc_update_dlist) from [<c0495e58>] (vc4_crtc_atomic_flush+0x10c/0x11c)
[  139.446758] [<c0495e58>] (vc4_crtc_atomic_flush) from [<c046a530>] (drm_atomic_helper_commit_planes+0x254/0x27c)
[  139.446812] [<c046a530>] (drm_atomic_helper_commit_planes) from [<c049baac>] (vc4_atomic_complete_commit+0x58/0xd4)
[  139.446858] [<c049baac>] (vc4_atomic_complete_commit) from [<c049bcc4>] (vc4_atomic_commit+0x19c/0x1bc)
[  139.446918] [<c049bcc4>] (vc4_atomic_commit) from [<c0485278>] (drm_atomic_commit+0x54/0x68)
[  139.446964] [<c0485278>] (drm_atomic_commit) from [<c046c890>] (restore_fbdev_mode+0x110/0x268)
[  139.447010] [<c046c890>] (restore_fbdev_mode) from [<c046ef3c>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x3c/0x80)
[  139.447051] [<c046ef3c>] (drm_fb_helper_restore_fbdev_mode_unlocked) from [<c046efd0>] (drm_fb_helper_set_par+0x50/0x6c)
[  139.447091] [<c046efd0>] (drm_fb_helper_set_par) from [<c046eeec>] (drm_fb_helper_hotplug_event+0xa0/0xb4)
[  139.447133] [<c046eeec>] (drm_fb_helper_hotplug_event) from [<c046fa10>] (drm_fbdev_cma_hotplug_event+0x20/0x24)
[  139.447176] [<c046fa10>] (drm_fbdev_cma_hotplug_event) from [<c049bd24>] (vc4_output_poll_changed+0x20/0x24)
[  139.447221] [<c049bd24>] (vc4_output_poll_changed) from [<c0460850>] (drm_kms_helper_hotplug_event+0x34/0x38)
[  139.447263] [<c0460850>] (drm_kms_helper_hotplug_event) from [<c0460cac>] (output_poll_execute+0x124/0x15c)
[  139.447310] [<c0460cac>] (output_poll_execute) from [<c013af68>] (process_one_work+0x328/0x674)
[  139.447352] [<c013af68>] (process_one_work) from [<c013c034>] (worker_thread+0x2d0/0x444)
[  139.447401] [<c013c034>] (worker_thread) from [<c0142078>] (kthread+0x14c/0x16c)
[  139.447454] [<c0142078>] (kthread) from [<c0108330>] (ret_from_fork+0x14/0x24)
[  139.447486] ---[ end trace 3d08db11084b457b ]---
```

lategoodbye, 2017-08-07:
In the bootlog of my RPI3 (32bit) with linux-next-20170807 i've seen this warning:

```
[   31.254715] ------------[ cut here ]------------
[   31.254748] WARNING: CPU: 1 PID: 38 at drivers/gpu/drm/drm_atomic_helper.c:1244 drm_atomic_helper_wait_for_vblanks.part.0+0x250/0x25c
[   31.254754] [CRTC:67:crtc-2] vblank wait timed out
[   31.254758] Modules linked in: vc4 snd_soc_core brcmfmac ac97_bus snd_pcm_dmaengine snd_pcm cfg80211 brcmutil snd_timer snd soundcore crc32_arm_ce
[   31.254809] CPU: 1 PID: 38 Comm: kworker/1:1 Not tainted 4.13.0-rc3-next-20170807 #1
[   31.254814] Hardware name: Generic DT based system
[   31.254826] Workqueue: events output_poll_execute
[   31.254849] [<c03119c4>] (unwind_backtrace) from [<c030ca20>] (show_stack+0x10/0x14)
[   31.254864] [<c030ca20>] (show_stack) from [<c0ce0d64>] (dump_stack+0x88/0xa4)
[   31.254881] [<c0ce0d64>] (dump_stack) from [<c0343460>] (__warn+0xd8/0x104)
[   31.254898] [<c0343460>] (__warn) from [<c03434c4>] (warn_slowpath_fmt+0x38/0x48)
[   31.254916] [<c03434c4>] (warn_slowpath_fmt) from [<c0811e3c>] (drm_atomic_helper_wait_for_vblanks.part.0+0x250/0x25c)
[   31.254975] [<c0811e3c>] (drm_atomic_helper_wait_for_vblanks.part.0) from [<bf279118>] (vc4_atomic_complete_commit+0x70/0xc0 [vc4])
[   31.255059] [<bf279118>] (vc4_atomic_complete_commit [vc4]) from [<bf279230>] (vc4_atomic_commit+0xc8/0x114 [vc4])
[   31.255108] [<bf279230>] (vc4_atomic_commit [vc4]) from [<c08141f8>] (restore_fbdev_mode_atomic+0x164/0x1d0)
[   31.255124] [<c08141f8>] (restore_fbdev_mode_atomic) from [<c0816dfc>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x40/0x90)
[   31.255137] [<c0816dfc>] (drm_fb_helper_restore_fbdev_mode_unlocked) from [<c0816c84>] (drm_fb_helper_set_par+0x2c/0x5c)
[   31.255151] [<c0816c84>] (drm_fb_helper_set_par) from [<c0816d7c>] (drm_fb_helper_hotplug_event.part.7+0xc8/0xe8)
[   31.255165] [<c0816d7c>] (drm_fb_helper_hotplug_event.part.7) from [<c0808b54>] (output_poll_execute+0x68/0x16c)
[   31.255181] [<c0808b54>] (output_poll_execute) from [<c035d884>] (process_one_work+0x154/0x47c)
[   31.255198] [<c035d884>] (process_one_work) from [<c035dbe4>] (worker_thread+0x38/0x554)
[   31.255213] [<c035dbe4>] (worker_thread) from [<c03632b0>] (kthread+0x12c/0x16c)
[   31.255227] [<c03632b0>] (kthread) from [<c0307cc0>] (ret_from_fork+0x14/0x34)
[   31.255234] ---[ end trace ae225bc82d84e95d ]---
```

anholt, 2017-08-08:
Failed to reproduce on drm-misc-next.  Just to be sure, do you have the firmware loading the same DT as u-boot is giving to the kernel?

lategoodbye, 2017-08-09:
Hardware setup:

`RPI3 -- <HDMI> -- Logilink HD0006 HDMI Switch -- <HDMI> -- HP ZR2440w TFT`

Kernel: linux-next-20170807 (multi_v7_defconfig)
Firmware: 2017-04-05
Device tree dts from arm (32bit)
No u-boot

config.txt
```
device_tree=bcm2837-rpi-3-b.dtb
enable_uart=1
dtoverlay=pi3-disable-bt
```

It was only able to trigger the warning by using the HDMI switch. Unfortunately this happens only sometimes not everytime.

lategoodbye, 2019-01-20:
I wasn't able to reproduce any of these warnings for a while. So i think this is fixed.

closed 2019-01-20 by lategoodbye