anholt/linux #111: dma-assert on 4.13 when booting aarch64

nullr0ute, 2017-09-12:
Not sure if this is vc4, the USB driver or something else but this is new on aarch64 with 4.13

<pre>
[   68.453898] [drm:vc4_get_tiling_ioctl [vc4]] *ERROR* Failed to look up GEM BO 0
[   69.179580] dwc2 3f980000.usb: DMA-API: cpu touching an active dma mapped cacheline [cln=0x000000000014dac0]
[   69.190968] ------------[ cut here ]------------
[   69.195696] WARNING: CPU: 0 PID: 727 at lib/dma-debug.c:614 debug_dma_assert_idle+0x1f0/0x260
[   69.204364] Modules linked in: nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables vfat fat vc4 snd_soc_core ac97_bus snd_pcm_dmaengine snd_seq snd_seq_device snd_pcm snd_timer smsc95xx snd usbnet soundcore mii drm_kms_helper drm brcmfmac fb_sys_fops syscopyarea brcmutil sysfillrect cfg80211 sysimgblt rfkill bcm2835_wdt bcm2835_rng crc32_ce leds_gpio mmc_block sdhci_iproc dwc2 sdhci_pltfm udc_core sdhci
[   69.276782]  pwm_bcm2835 i2c_bcm2835 bcm2835 bcm2835_dma phy_generic
[   69.283308] CPU: 0 PID: 727 Comm: dhclient Not tainted 4.13.1-302.fc27.aarch64 #1
[   69.290910] Hardware name: raspberrypi rpi/rpi, BIOS 2017.09-rc2 08/25/2017
[   69.297985] task: ffff8000108c4e00 task.stack: ffff800010930000
[   69.304007] PC is at debug_dma_assert_idle+0x1f0/0x260
[   69.309234] LR is at debug_dma_assert_idle+0x1f0/0x260
[   69.314460] pc : [<ffff00000853f420>] lr : [<ffff00000853f420>] pstate: 00400145
[   69.321973] sp : ffff800010933a30
[   69.325344] x29: ffff800010933a30 x28: 0000000000000002 
[   69.330766] x27: ffff800038fe12a0 x26: 0000ffff98b86000 
[   69.336186] x25: ffff00000a17d880 x24: ffff0000090f3000 
[   69.341608] x23: 0000000000000140 x22: ffff000009186f80 
[   69.347029] x21: ffff00000a17d880 x20: ffff000009186000 
[   69.352450] x19: ffff80003b5fc600 x18: 0000000000000000 
[   69.357869] x17: 0000000000000000 x16: 0000000000000000 
[   69.363290] x15: 0000000000000000 x14: ffff8000150dfb78 
[   69.368710] x13: ffff000009187000 x12: ffff8000150dfb50 
[   69.374129] x11: 0000000000000003 x10: 0000000000002300 
[   69.379549] x9 : ffff8000109336d0 x8 : 0000000000000001 
[   69.384968] x7 : 0000000000000000 x6 : ffff80003df07ef0 
[   69.390388] x5 : ffff80003df07ef0 x4 : 0000000000000000 
[   69.395808] x3 : ffff80003df0e180 x2 : ffff8000108c4e00 
[   69.401228] x1 : ffff8000108c4e00 x0 : 0000000000000060 
[   69.406650] Call trace:
[   69.409145] Exception stack(0xffff800010933850 to 0xffff800010933980)
[   69.415694] 3840:                                   ffff80003b5fc600 0001000000000000
[   69.423655] 3860: 00000000021e4000 ffff00000853f420 0000000000400145 000000000000003d
[   69.431614] 3880: ffff00000a17d880 0000ffff98b86000 ffff800038fe12a0 0000000000000002
[   69.439574] 38a0: ffff8000109338d0 0000000000000000 ffff800010933a30 ffff800010933a30
[   69.447534] 38c0: ffff8000109339f0 00000000ffffffc8 ffff800010933910 ffff00000816296c
[   69.455494] 38e0: ffff000008dc8448 ffff8000109339b0 ffff800010933a30 ffff800010933a30
[   69.463454] 3900: 0000000000000060 ffff8000108c4e00 ffff8000108c4e00 ffff80003df0e180
[   69.471413] 3920: 0000000000000000 ffff80003df07ef0 ffff80003df07ef0 0000000000000000
[   69.479372] 3940: 0000000000000001 ffff8000109336d0 0000000000002300 0000000000000003
[   69.487332] 3960: ffff8000150dfb50 ffff000009187000 ffff8000150dfb78 0000000000000000
[   69.495293] [<ffff00000853f420>] debug_dma_assert_idle+0x1f0/0x260
[   69.501583] [<ffff0000082a38c8>] wp_page_copy+0xa0/0x828
[   69.506985] [<ffff0000082a586c>] do_wp_page+0x9c/0x5f8
[   69.512211] [<ffff0000082aa2a8>] __handle_mm_fault+0x378/0xc70
[   69.518142] [<ffff0000082aad54>] handle_mm_fault+0x1b4/0x2f8
[   69.523900] [<ffff000008ab3680>] do_page_fault+0x228/0x480
[   69.529483] [<ffff0000080813bc>] do_mem_abort+0x54/0xb0
[   69.534795] Exception stack(0xffff800010933cc0 to 0xffff800010933df0)
[   69.541345] 3cc0: 00000000000002d7 0001000000000000 00000000021e4000 ffff000008116a30
[   69.549305] 3ce0: ffff8000108c4e00 0000000000000000 ffff800010933d40 ffff00000814ab7c
[   69.557265] 3d00: 000000000000001e ffff8000108c5740 ffff8000108c4e00 0000000000000000
[   69.565224] 3d20: ffff8000108c4e00 0000000000000000 0000000000000000 0000000000000000
[   69.573184] 3d40: ffff800010933dd0 ffff00000814b978 ffff0000090c9000 ffff00000911eef0
[   69.581143] 3d60: ffff8000108c4e00 0000000000000000 ffff000009f4d000 0000000000000000
[   69.589101] 3d80: 0000000000000000 0000000000001000 0000000000000000 0000000000000000
[   69.597061] 3da0: 0000000000000001 ffff800010933ec0 0000000000002300 0000000000000003
[   69.605020] 3dc0: ffff8000150dc6d8 ffff000009187000 ffff8000150dc800 0000000000000000
[   69.612981] 3de0: 0000000000000000 0000000000517b1a
[   69.617950] [<ffff000008082df8>] el1_da+0x24/0xa0
[   69.622744] [<ffff000008083584>] ret_from_fork+0x4/0x40
[   69.628058] ---[ end trace b63f5ca203fab10f ]---
[   69.633852] Mapped at:
[   69.637298]  dma_entry_alloc+0x68/0xa8
[   69.641829]  debug_dma_alloc_coherent+0x54/0xb0
[   69.646671]  dwc2_hcd_init+0x5fc/0x7c0 [dwc2]
[   69.651418]  dwc2_driver_probe+0x4d0/0x5b8 [dwc2]
[   69.656616]  platform_drv_probe+0x60/0xc0
[   70.412273] [drm:vc4_get_tiling_ioctl [vc4]] *ERROR* Failed to look up GEM BO 0
</pre>

lategoodbye, 2017-09-12:
@nullr0ute Please provide all enabled settings under "Kernel hacking" of your kernel config.

nullr0ute, 2017-09-12:
The build config is attached. Not sure what you mean by the "Kernel hacking" bit, I don't see it referenced in the wiki here can you provide a link?

[config-4.13.1-302.fc27.armv7hl.txt](https://github.com/anholt/linux/files/1296828/config-4.13.1-302.fc27.armv7hl.txt)

nullr0ute, 2017-09-12:
Actually apologies, that was the 32 bit kernel build, this is the 64 bit config
[config-4.13.1-302.fc27.aarch64.txt](https://github.com/anholt/linux/files/1296841/config-4.13.1-302.fc27.aarch64.txt)

lategoodbye, 2017-09-12:
"Kernel hacking" is a section within the kernel config, but the provided links are sufficient. Thanks

lategoodbye, 2017-09-12:
Could you please try this [patch](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20170912&id=723288836628bc1c0855f3bb7b64b1803e4b9e4a)?

nullr0ute, 2017-09-12:
Looks much better with that patch

lategoodbye, 2017-09-13:
Does it mean the patch fixes the issue?

nullr0ute, 2017-09-13:
Yep, with further testing I've not seen it show up again

lategoodbye, 2017-09-13:
AFAIK, this patch would land in 4.14.

nullr0ute, 2017-09-19:
I don't think this is completely fixed, I'm still seeing a very similar crash with that patch applied to 4.13.2

<pre>
[   60.017128] vc4-drm soc:gpu: DMA-API: cpu touching an active dma mapped cacheline [cln=0x0000000000194b80]
[   60.046053] ------------[ cut here ]------------
[   60.068318] WARNING: CPU: 0 PID: 468 at lib/dma-debug.c:614 debug_dma_assert_idle+0x1f0/0x260
[   60.094849] Modules linked in: vfat fat smsc95xx usbnet mii vc4 snd_soc_core ac97_bus snd_pcm_dmaengine snd_seq snd_seq_device brcmfmac snd_pcm brcmutil cfg80211 rfkill snd_timer snd soundcore drm_kms_helper drm crc32_ce fb_sys_fops syscopyarea sysfillrect sysimgblt bcm2835_rng bcm2835_wdt leds_gpio mmc_block sdhci_iproc dwc2 sdhci_pltfm udc_core sdhci pwm_bcm2835 i2c_bcm2835 bcm2835 bcm2835_dma i2c_dev
[   60.169691] CPU: 0 PID: 468 Comm: firewalld Not tainted 4.13.2-300.fc27.aarch64 #1
[   60.196097] Hardware name: raspberrypi rpi/rpi, BIOS 2017.09-rc4 09/06/2017
[   60.221262] task: ffff80002625ce00 task.stack: ffff800024ab4000
[   60.245411] PC is at debug_dma_assert_idle+0x1f0/0x260
[   60.268002] LR is at debug_dma_assert_idle+0x1f0/0x260
[   60.289809] pc : [<ffff00000853e418>] lr : [<ffff00000853e418>] pstate: 00400145
[   60.313534] sp : ffff800024ab7b90
[   60.332425] x29: ffff800024ab7b90 x28: 0000000000000002
[   60.352873] x27: ffff800027fd0520 x26: 0000ffff99150000
[   60.372698] x25: ffff00000a17e880 x24: ffff0000090f3000
[   60.392312] x23: 0000000000000140 x22: ffff000009186f80
[   60.411420] x21: ffff00000a17e880 x20: ffff000009186000
[   60.429897] x19: ffff80002a605a00 x18: 0000000000000010
[   60.447760] x17: 0000000000000000 x16: 0000000000000000
[   60.464950] x15: 0000000000000001 x14: ffff000000bbd038
[   60.482126] x13: 0000000000000000 x12: 0000000000000002
[   60.499197] x11: ffff0000092fdb58 x10: 0000000000002300
[   60.516192] x9 : ffff800024ab7830 x8 : 0000000000000001
[   60.533142] x7 : 0000000000000000 x6 : ffff80003e507ef0
[   60.550024] x5 : ffff80003e507ef0 x4 : 0000000000000000
[   60.566842] x3 : ffff80003e50e180 x2 : ffff80002625ce00
[   60.583617] x1 : ffff80002625ce00 x0 : 000000000000005e
[   60.600350] Call trace:
[   60.614058] Exception stack(0xffff800024ab79b0 to 0xffff800024ab7ae0)
[   60.632102] 79a0:                                   ffff80002a605a00 0001000000000000
[   60.651654] 79c0: 00000000021e5000 ffff00000853e418 0000000000400145 000000000000003d
[   60.671252] 79e0: ffff00000a17e880 0000ffff99150000 ffff800027fd0520 0000000000000002
[   60.690871] 7a00: 0000000000000001 0000000000000000 ffff800024ab7b90 ffff800024ab7b90
[   60.710487] 7a20: ffff800024ab7b50 00000000ffffffc8 ffff800024ab7a70 ffff00000816296c
[   60.730110] 7a40: ffff000008dc9448 ffff800024ab7b10 ffff800024ab7b90 ffff800024ab7b90
[   60.749707] 7a60: 000000000000005e ffff80002625ce00 ffff80002625ce00 ffff80003e50e180
[   60.769339] 7a80: 0000000000000000 ffff80003e507ef0 ffff80003e507ef0 0000000000000000
[   60.788957] 7aa0: 0000000000000001 ffff800024ab7830 0000000000002300 ffff0000092fdb58
[   60.808623] 7ac0: 0000000000000002 0000000000000000 ffff000000bbd038 0000000000000001
[   60.828336] [<ffff00000853e418>] debug_dma_assert_idle+0x1f0/0x260
[   60.846434] [<ffff0000082a38c8>] wp_page_copy+0xa0/0x828
[   60.863625] [<ffff0000082a586c>] do_wp_page+0x9c/0x5f8
[   60.880681] [<ffff0000082aa2a8>] __handle_mm_fault+0x378/0xc70
[   60.898477] [<ffff0000082aad54>] handle_mm_fault+0x1b4/0x2f8
[   60.916414] [<ffff000008ab47a0>] do_page_fault+0x228/0x480
[   60.934041] [<ffff0000080813bc>] do_mem_abort+0x54/0xb0
[   60.951282] Exception stack(0xffff800024ab7e20 to 0xffff800024ab7f50)
[   60.969854] 7e20: 0000000000000000 000080003543a000 ffffffffffffffff 0000ffffa2513108
[   60.990018] 7e40: 0000000000000005 ffff80002a28df80 ffff80002a28dfd8 ffff80002a28df80
[   61.010205] 7e60: ffff800024ab7ea0 ffff000008309ac0 0000000000000005 000080003543a000
[   61.030458] 7e80: ffff800024ab7eb0 ffff00000814a5f8 ffff800024ab7ec0 000080003543a000
[   61.050683] 7ea0: ffffffffffffffff ffff000008146348 ffff80003e507070 ffff0000080833b8
[   61.070917] 7ec0: 0000ffff99132e30 0000000000031740 0000ffff9914e5a8 0000ffff99126230
[   61.091175] 7ee0: 0000ffff99126230 0000ffff99127b20 0000ffff991222ff 0000000000000000
[   61.111422] 7f00: 00000000066d1f10 0000ffff991225e0 0000000000000018 0000000000005520
[   61.131636] 7f20: 000000000000000b 0000ffffa252b000 fffffffffffffffe 0000000000000000
[   61.151850] 7f40: 0000ffffa2546f90 0000ffffa1f81580
[   61.168980] [<ffff0000080831dc>] el0_da+0x20/0x24
[   61.185952] ---[ end trace bacab57ab194a518 ]---
[   61.203373] Mapped at:
[   61.218285]  dma_entry_alloc+0x68/0xa8
[   61.234829]  debug_dma_alloc_coherent+0x54/0xb0
[   61.252367]  drm_gem_cma_create+0xdc/0x148 [drm]
[   61.269803]  drm_fbdev_cma_create+0x9c/0x348 [drm_kms_helper]
[   61.288470]  drm_fb_helper_initial_config+0x1f8/0x448 [drm_kms_helper]
</pre>

nullr0ute, 2017-09-19:
With a couple of reboots I see the Mapped at: change around a little. EG:

[   66.104625] Mapped at:
[   66.107660]  dma_entry_alloc+0x68/0xa8
[   66.111718]  debug_dma_alloc_coherent+0x54/0xb0
[   66.118114]  snd_dma_alloc_pages+0x178/0x200 [snd_pcm]
[   66.123851]  snd_pcm_lib_preallocate_pages1+0x118/0x1e8 [snd_pcm]
[   66.130524]  snd_pcm_lib_preallocate_pages+0x48/0x60 [snd_pcm]

lategoodbye, 2017-09-20:
Could you please test it with 4.14-rc1?

lategoodbye, 2017-09-23:
If it's reproducible please report to:

```
Christoph Hellwig <hch@lst.de>
Marek Szyprowski <m.szyprowski@samsung.com>
Robin Murphy <robin.murphy@arm.com>
iommu@lists.linux-foundation.org
linux-rpi-kernel@lists.infradead.org
linux-arm-kernel@lists.infradead.org
```

lategoodbye, 2019-01-20:
@nullr0ute Is this issue still reproducible with Fedora 29?

nullr0ute, 2019-01-21:
I believe this one is fixed, I don't remember seeing it recently.

closed 2019-01-21 by lategoodbye