anholt/linux #112: Crash in vc4_hdmi_encoder_enable on boot with 4.13.x

nullr0ute, 2017-09-17:
Fedora 27 with 4.13.2, mesa 7.2.0 and gnome 3.26 gets this crash on boot shortly after loading the vc4 module. Happens twice but then not again, although we then get crashes in drm_atomic_helper_wait_for_vblanks (will do a separate report for that)

<pre>
[   52.354172] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
[   52.354206] vc4_hdmi 3f902000.hdmi: ASoC: no DMI vendor name!
[   52.363308] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   52.363727] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[   52.363885] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   52.364348] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   52.364763] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   52.365149] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   52.441958] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   52.441977] checking generic (3e314000 6d8c00) vs hw (0 ffffffff)
[   52.441983] fb: switching to vc4drmfb from simple
[   52.601810] Console: switching to colour dummy device 80x30
[   52.621136] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   52.621293] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   52.621385] [drm] Driver supports precise vblank timestamp query.
[   53.692614] ------------[ cut here ]------------
[   53.692730] WARNING: CPU: 1 PID: 461 at drivers/gpu/drm/vc4/vc4_hdmi.c:658 vc4_hdmi_encoder_enable+0x21c/0xabc [vc4]
[   53.692736] Timeout waiting for VC4_HDMI_SCHEDULER_CONTROL_HDMI_ACTIVE
[   53.692741] Modules linked in: vc4(+) snd_soc_core ac97_bus snd_pcm_dmaengine snd_seq snd_seq_device snd_pcm hci_uart brcmfmac btbcm btqca brcmutil btintel snd_timer cfg80211 bluetooth snd soundcore drm_kms_helper ecdh_generic rfkill drm fb_sys_fops syscopyarea sysfillrect sysimgblt bcm2835_wdt bcm2835_thermal bcm2835_rng leds_gpio smsc95xx usbnet mii mmc_block sdhci_iproc sdhci_pltfm crc32_arm_ce dwc2 udc_core sdhci pwm_bcm2835 bcm2835 i2c_bcm2835 bcm2835_dma phy_generic
[   53.692950] CPU: 1 PID: 461 Comm: systemd-udevd Not tainted 4.13.2-300.fc27.armv7hl #1
[   53.692956] Hardware name: Generic DT based system
[   53.692989] [<c0312af4>] (unwind_backtrace) from [<c030d264>] (show_stack+0x18/0x1c)
[   53.693007] [<c030d264>] (show_stack) from [<c0b66174>] (dump_stack+0xa0/0xd8)
[   53.693024] [<c0b66174>] (dump_stack) from [<c034f258>] (__warn+0xe4/0x104)
[   53.693039] [<c034f258>] (__warn) from [<c034f2b4>] (warn_slowpath_fmt+0x3c/0x4c)
[   53.693131] [<c034f2b4>] (warn_slowpath_fmt) from [<bf5f939c>] (vc4_hdmi_encoder_enable+0x21c/0xabc [vc4])
[   53.693346] [<bf5f939c>] (vc4_hdmi_encoder_enable [vc4]) from [<bf1c186c>] (drm_atomic_helper_commit_modeset_enables+0x158/0x180 [drm_kms_helper])
[   53.693552] [<bf1c186c>] (drm_atomic_helper_commit_modeset_enables [drm_kms_helper]) from [<bf5f6750>] (vc4_atomic_complete_commit+0x5c/0xbc [vc4])
[   53.693720] [<bf5f6750>] (vc4_atomic_complete_commit [vc4]) from [<bf5f6930>] (vc4_atomic_commit+0x180/0x190 [vc4])
[   53.693922] [<bf5f6930>] (vc4_atomic_commit [vc4]) from [<bf1c6ccc>] (restore_fbdev_mode+0x94/0x24c [drm_kms_helper])
[   53.694159] [<bf1c6ccc>] (restore_fbdev_mode [drm_kms_helper]) from [<bf1c9148>] (drm_fb_helper_restore_fbdev_mode_unlocked+0x38/0x70 [drm_kms_helper])
[   53.694398] [<bf1c9148>] (drm_fb_helper_restore_fbdev_mode_unlocked [drm_kms_helper]) from [<bf1c91d4>] (drm_fb_helper_set_par+0x54/0x64 [drm_kms_helper])
[   53.694528] [<bf1c91d4>] (drm_fb_helper_set_par [drm_kms_helper]) from [<c077715c>] (fbcon_init+0x2cc/0x454)
[   53.694548] [<c077715c>] (fbcon_init) from [<c07f57e4>] (visual_init+0xc8/0x118)
[   53.694563] [<c07f57e4>] (visual_init) from [<c07f7178>] (do_bind_con_driver+0x26c/0x2dc)
[   53.694579] [<c07f7178>] (do_bind_con_driver) from [<c07f7560>] (do_take_over_console+0x174/0x1a8)
[   53.694595] [<c07f7560>] (do_take_over_console) from [<c077733c>] (do_fbcon_takeover+0x58/0xc0)
[   53.694612] [<c077733c>] (do_fbcon_takeover) from [<c03785e0>] (notifier_call_chain+0x84/0xb0)
[   53.694629] [<c03785e0>] (notifier_call_chain) from [<c0378bb4>] (__blocking_notifier_call_chain+0x48/0x60)
[   53.694645] [<c0378bb4>] (__blocking_notifier_call_chain) from [<c0378be8>] (blocking_notifier_call_chain+0x1c/0x24)
[   53.694660] [<c0378be8>] (blocking_notifier_call_chain) from [<c078079c>] (register_framebuffer+0x234/0x278)
[   53.694788] [<c078079c>] (register_framebuffer) from [<bf1c8e7c>] (drm_fb_helper_initial_config+0x1c0/0x3a8 [drm_kms_helper])
[   53.695028] [<bf1c8e7c>] (drm_fb_helper_initial_config [drm_kms_helper]) from [<bf1c9810>] (drm_fbdev_cma_init_with_funcs+0xb8/0xf4 [drm_kms_helper])
[   53.695231] [<bf1c9810>] (drm_fbdev_cma_init_with_funcs [drm_kms_helper]) from [<bf5f6afc>] (vc4_kms_load+0xd0/0x108 [vc4])
[   53.695398] [<bf5f6afc>] (vc4_kms_load [vc4]) from [<bf5f26b4>] (vc4_drm_bind+0xe4/0x128 [vc4])
[   53.695493] [<bf5f26b4>] (vc4_drm_bind [vc4]) from [<c0838fdc>] (try_to_bring_up_master+0x1ec/0x254)
[   53.695512] [<c0838fdc>] (try_to_bring_up_master) from [<c0839628>] (component_master_add_with_match+0xc0/0xec)
[   53.695604] [<c0839628>] (component_master_add_with_match) from [<bf5f276c>] (vc4_platform_drm_probe+0x74/0xb0 [vc4])
[   53.695698] [<bf5f276c>] (vc4_platform_drm_probe [vc4]) from [<c0840f34>] (platform_drv_probe+0x58/0xa4)
[   53.695714] [<c0840f34>] (platform_drv_probe) from [<c083eb40>] (driver_probe_device+0x28c/0x424)
[   53.695728] [<c083eb40>] (driver_probe_device) from [<c083ed68>] (__driver_attach+0x90/0x100)
[   53.695744] [<c083ed68>] (__driver_attach) from [<c083ccd8>] (bus_for_each_dev+0x84/0x94)
[   53.695763] [<c083ccd8>] (bus_for_each_dev) from [<c083df4c>] (bus_add_driver+0x1bc/0x234)
[   53.695779] [<c083df4c>] (bus_add_driver) from [<c083ff6c>] (driver_register+0xa8/0xe8)
[   53.695794] [<c083ff6c>] (driver_register) from [<c0301de8>] (do_one_initcall+0x130/0x158)
[   53.695812] [<c0301de8>] (do_one_initcall) from [<c040375c>] (do_init_module+0x60/0x1dc)
[   53.695829] [<c040375c>] (do_init_module) from [<c0402c0c>] (load_module+0x2430/0x24d8)
[   53.695846] [<c0402c0c>] (load_module) from [<c0402f24>] (SyS_finit_module+0xb0/0xc4)
[   53.695863] [<c0402f24>] (SyS_finit_module) from [<c03081a8>] (__sys_trace_return+0x0/0x10)
[   53.695911] ---[ end trace b2e9be4e1ce19770 ]---
[   53.695916] ------------[ cut here ]------------
[   53.696007] WARNING: CPU: 1 PID: 461 at drivers/gpu/drm/vc4/vc4_hdmi.c:677 vc4_hdmi_encoder_enable+0x904/0xabc [vc4]
[   53.696012] Modules linked in: vc4(+) snd_soc_core ac97_bus snd_pcm_dmaengine snd_seq snd_seq_device snd_pcm hci_uart brcmfmac btbcm btqca brcmutil btintel snd_timer cfg80211 bluetooth snd soundcore drm_kms_helper ecdh_generic rfkill drm fb_sys_fops syscopyarea sysfillrect sysimgblt bcm2835_wdt bcm2835_thermal bcm2835_rng leds_gpio smsc95xx usbnet mii mmc_block sdhci_iproc sdhci_pltfm crc32_arm_ce dwc2 udc_core sdhci pwm_bcm2835 bcm2835 i2c_bcm2835 bcm2835_dma phy_generic
[   53.696204] CPU: 1 PID: 461 Comm: systemd-udevd Tainted: G        W       4.13.2-300.fc27.armv7hl #1
[   53.696209] Hardware name: Generic DT based system
[   53.696228] [<c0312af4>] (unwind_backtrace) from [<c030d264>] (show_stack+0x18/0x1c)
[   53.696243] [<c030d264>] (show_stack) from [<c0b66174>] (dump_stack+0xa0/0xd8)
[... 13 lines cut ...]
[   53.697799] [<c07f7560>] (do_take_over_console) from [<c077733c>] (do_fbcon_takeover+0x58/0xc0)
[   53.697815] [<c077733c>] (do_fbcon_takeover) from [<c03785e0>] (notifier_call_chain+0x84/0xb0)
[   53.697830] [<c03785e0>] (notifier_call_chain) from [<c0378bb4>] (__blocking_notifier_call_chain+0x48/0x60)
[   53.697846] [<c0378bb4>] (__blocking_notifier_call_chain) from [<c0378be8>] (blocking_notifier_call_chain+0x1c/0x24)
[   53.697861] [<c0378be8>] (blocking_notifier_call_chain) from [<c078079c>] (register_framebuffer+0x234/0x278)
[   53.697985] [<c078079c>] (register_framebuffer) from [<bf1c8e7c>] (drm_fb_helper_initial_config+0x1c0/0x3a8 [drm_kms_helper])
[   53.698224] [<bf1c8e7c>] (drm_fb_helper_initial_config [drm_kms_helper]) from [<bf1c9810>] (drm_fbdev_cma_init_with_funcs+0xb8/0xf4 [drm_kms_helper])
[   53.698427] [<bf1c9810>] (drm_fbdev_cma_init_with_funcs [drm_kms_helper]) from [<bf5f6afc>] (vc4_kms_load+0xd0/0x108 [vc4])
[   53.698594] [<bf5f6afc>] (vc4_kms_load [vc4]) from [<bf5f26b4>] (vc4_drm_bind+0xe4/0x128 [vc4])
[... cut ...]
[   53.698979] [<c083ff6c>] (driver_register) from [<c0301de8>] (do_one_initcall+0x130/0x158)
[   53.698995] [<c0301de8>] (do_one_initcall) from [<c040375c>] (do_init_module+0x60/0x1dc)
[   53.699010] [<c040375c>] (do_init_module) from [<c0402c0c>] (load_module+0x2430/0x24d8)
[   53.699026] [<c0402c0c>] (load_module) from [<c0402f24>] (SyS_finit_module+0xb0/0xc4)
[   53.699042] [<c0402f24>] (SyS_finit_module) from [<c03081a8>] (__sys_trace_return+0x0/0x10)
[   53.699085] ---[ end trace b2e9be4e1ce19771 ]---
</pre>

lategoodbye, 2019-01-20:
@nullr0ute Is this issue still reproducible with Fedora 29?

nullr0ute, 2019-01-21:
I think we can close it off, I don't remember seeing it, can always re-open.

closed 2019-01-21 by lategoodbye