anholt/linux #114: vc4: NULL pointer dereference in 4.9.51-v7+

mattyway, 2017-09-29:
I'm trying to use the VC4 driver on a Raspberry Pi 3. I have updated the kernel via rpi-update and installed the snapd package, but otherwise it should be a regular Raspbian image.

After enabling the driver via raspi-config and rebooting, it seems to be crashing on a NULL pointer dereference. It does reboot successfully sometimes, but a second reboot will trigger the same crash.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.51-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1036 SMP Fri Sep 22 19:49:07 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @b7754000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: 8250.nr_uarts=1 cma=256M bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=d3646955-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 633128K/917504K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 22232K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xb8800000 - 0xff800000   (1136 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xb8000000   ( 896 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79034   ( 485 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3daa4   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000304] Console: colour dummy device 80x30
[    0.001216] console [tty1] enabled
[    0.001265] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001334] pid_max: default: 32768 minimum: 301
[    0.001667] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001709] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002750] Disabling memory control group subsystem
[    0.002854] CPU: Testing write buffer coherency: ok
[    0.002919] ftrace: allocating 22401 entries in 66 pages
[    0.049827] CPU0: update cpu_capacity 1024
[    0.049879] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.049941] Setting up static identity map for 0x100000 - 0x100034
[    0.051853] CPU1: update cpu_capacity 1024
[    0.051859] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.052551] CPU2: update cpu_capacity 1024
[    0.052557] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.053236] CPU3: update cpu_capacity 1024
[    0.053243] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.053330] Brought up 4 CPUs
[    0.053505] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.053535] CPU: All CPU(s) started in HYP mode.
[    0.053561] CPU: Virtualization extensions available.
[    0.054381] devtmpfs: initialized
[    0.065764] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.066079] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.066142] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.076725] pinctrl core: initialized pinctrl subsystem
[    0.077762] NET: Registered protocol family 16
[    0.080162] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.089024] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.089074] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.089250] Serial: AMBA PL011 UART driver
[... 319 lines cut ...]
[    9.963247] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.971540] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 0001367f 00013680
[    9.979842] [<8013d564>] (kthread_data) from [<801379d4>] (wq_worker_sleeping+0x18/0xd8)
[    9.988055] [<801379d4>] (wq_worker_sleeping) from [<807163bc>] (__schedule+0x51c/0x7bc)
[    9.996267] [<807163bc>] (__schedule) from [<80149700>] (do_task_dead+0x98/0x9c)
[   10.003786] [<80149700>] (do_task_dead) from [<80121788>] (do_exit+0x754/0xaec)
[   10.011206] [<80121788>] (do_exit) from [<8010c2b0>] (die+0x254/0x440)
[   10.017828] [<8010c2b0>] (die) from [<80118eb4>] (__do_kernel_fault.part.0+0x74/0x1f4)
[   10.025860] [<80118eb4>] (__do_kernel_fault.part.0) from [<8071bbc8>] (do_page_fault+0x250/0x3bc)
[   10.034861] [<8071bbc8>] (do_page_fault) from [<8071bdf0>] (do_translation_fault+0xbc/0xc0)
[   10.043333] [<8071bdf0>] (do_translation_fault) from [<801011e8>] (do_DataAbort+0x48/0xc4)
[   10.051715] [<801011e8>] (do_DataAbort) from [<8071b174>] (__dabt_svc+0x54/0x80)
[   10.059212] Exception stack(0xb3e9de70 to 0xb3e9deb8)
[   10.064331] de60:                                     00080000 20000013 00000002 00000002
[   10.072624] de80: b3e5b118 00000001 00000000 b3e5b0e0 00000000 00000008 00000000 b3e9dedc
[   10.080915] dea0: b3e9dea8 b3e9dec0 8071acec 7f4a8760 60000093 ffffffff
[   10.087659] [<8071b174>] (__dabt_svc) from [<7f4a8760>] (vc4_overflow_mem_work+0x80/0x120 [vc4])
[   10.096599] [<7f4a8760>] (vc4_overflow_mem_work [vc4]) from [<80136720>] (process_one_work+0x158/0x454)
[   10.106130] [<80136720>] (process_one_work) from [<80136a80>] (worker_thread+0x64/0x568)
[   10.114337] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
[   10.121839] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
[   10.129164] Code: e24cb004 e52de004 e8bd4000 e59033f8 (e5130014)
[   10.135342] ---[ end trace 1773a549536bf2f7 ]---
[   10.140022] Fixing recursive fault but reboot is needed!
```

lategoodbye, 2017-09-30:
@mattyway Could you try this patch https://gist.github.com/lategoodbye/73f489adacb2e689ccc7017397c90a73 ?

lategoodbye, 2017-10-04:
@mattyway Any chance to test the linked patch?

anyc, 2017-10-08:
Unfortunately, the oops still occurs with the patch. :/

anyc, 2017-10-08:
By the way, I don't know if it was the same cause but I had freezing RPis also with earlier 4.9 kernels - less than 4.9.40 I think.

lategoodbye, 2017-10-08:
@anyc I'm not sure your issue is related to this one.

lategoodbye, 2017-11-11:
@mattyway 

Please try the following patches (keep in mind they are written for current mainline, so maybe they need rework):
https://patchwork.kernel.org/patch/10052207/
https://patchwork.kernel.org/patch/10052209/

anholt, 2017-11-14:
Pushed to drm-misc-fixes:

```
commit 253696ccd613fbdaa5aba1de44c461a058e0a114 (HEAD -> drm-misc-fixes, drm-misc/for-linux-next-fixes, drm-misc/drm-misc-fixes)
Author: Stefan Schake <stschake@gmail.com>
Date:   Fri Nov 10 02:05:06 2017 +0100

    drm/vc4: Account for interrupts in flight
```

closed 2017-11-14 by anholt