anholt/linux #115: Kernel oops in vc4_overflow_mem_work

anyc, 2017-10-15:
I get exactly the same kernel oops as described in #114 but at a random time after boot. I'd say around 30 minutes to several hours after booting the device. I discussed this issue already with @lategoodbye in my original report at https://github.com/raspberrypi/linux/issues/2217. I try to summarize it here:

* I use a self-compiled kernel but it also occurs with a raspbian image
* also occurs with kernel 4.9.36 and another RPi2 and a RPi3
* I use the RPi for a [slideshow](https://github.com/anyc/qante) with large photographs but it also happens without the app
* I see a lot of "Resetting GPU" messages and the oops seems to occur  ~50 ms after one of those messages
* I do not own an official power supply but I used two different, "non-cheap" supplies with different cables

If it helps, I can also send you a link to the SD card image.

anyc, 2017-10-26:
The same also happens with an original RPi power supply.
The last output from a custom surveillance script calling vcgencmd every 10s:
```
frequency(1)=250000000
frequency(45)=600000000
temp=43.3'C
throttled=0x0
```

lategoodbye, 2017-10-29:
First step to fix this issue, would be a reproducible scenario which also works for us.

Is there a chance to reproduce this issue with a default raspbian without any custom applications?

anyc, 2017-10-29:
Right now raspbian is running for 9h without the app and without oops. I'll let it run for a little longer.

I can provide you the raspbian image with just the app and pictures. I just don't want to share them publicly in original resolution.

Thank you!

anyc, 2017-11-10:
I tried different runs during the last days but the results are all the same. Raspbian without the app works. Both images cause an oops after a random amount of time with the app running.

lategoodbye, 2017-11-11:
@anyc 

Please try the following patches (keep in mind they are written for current mainline, so maybe they need rework):
https://patchwork.kernel.org/patch/10052207/
https://patchwork.kernel.org/patch/10052209/

anyc, 2017-11-11:
It looks good so far. No oops after 8h with the app running.

anholt, 2017-11-14:
I've merged the patch for upstream and sent the PR for 4.9 downstream.  If your testing says the patch fixed it, please close.  (and thanks for testing!)

anyc, 2018-01-07:
So, after several weeks there was no reported issue. Thank you!

closed 2018-01-07 by anyc