anholt/linux #116: vc4: HDMI detection fail on some monitors

babyimp, 2017-10-17:
Hi,

Android Things DP 5.1 installed on RPI3 fails to detect HDMI connection status on some monitors such as UOTOO 7" TFT monitor and PI-topCEED (all monitors are physically connected via HDMI).

I traced where to fail and notice that a function vc4_hdmi_connector_detect in vc4_hdmi.c returns connector_status_disconnected in such cases.

Any hints or ideas to solve this issue?

anholt, 2017-10-18:
What kernel are you using?  Does the failing monitor not expose an EDID block?

In general, when reporting display issues, you need drm.debug=0x1e on the kernel command line and to attach (not paste) the dmesg.

babyimp, 2017-10-19:
Sorry about not recognizing the posting rule.

Kernel is 4.4. The monitor exposes an EDID block.
dmesgs both on success case and on fail case attached.
[dmesg.tar.gz](https://github.com/anholt/linux/files/1396770/dmesg.tar.gz)

anholt, 2017-10-19:
4.4 is ancient and I don't support it any more.  Notably, it doesn't have HDMI HPD support on the pi3.  Please update to the current supported branch (4.9) if you're going to stay on the downstream kernel tree.

babyimp, 2017-10-23:
got it. let's see what will happen after updating to 4.9.

anholt, 2017-11-09:
Can we close this?

babyimp, 2017-11-09:
oh, sorry. issue closed.

babyimp, 2018-01-04:
I still see this issue after upgrading to 4.9. no issues with HP monitor, Acer monitor, and some others, while edid is not retrieved from UOTOO "7 TFT and some.
dmesg attached (I also turn on i2c debug).
@anholt can you give some comments?
[hdmi15.txt](https://github.com/anholt/linux/files/1602095/hdmi15.txt)

anholt, 2018-02-05:
You've got a timeout on I2C.  This may be your monitor being bad at I2C, or bcm2835-i2c being bad at I2C.   I've got one fix laying around on the `bcm2835-i2c-clock-stretch` branch, and I think we also have some potential issues due to not setting I2C_DEL.

lategoodbye, 2018-02-05:
Do you mean the `drm-vc4-i2c-fix` branch?
Should we send this patch to upstream?

anholt, 2018-02-06:
Oh, I hadn't pushed the rename of the branch.  Yeah, should probably submit.

anholt, 2018-02-06:
https://patchwork.kernel.org/patch/9148267/ was another i2c patch I wrote at one point

anholt, 2018-02-22:
The delay patch should now be applied to the i2c tree, and it fixed EDID at least for Boris.  https://patchwork.ozlabs.org/patch/870884/

anholt, 2019-01-17:
@babyimp have you tried since the i2c fixes?