anholt/linux #118: kernel driver vc4 with cec support, TV remote control not working

nkichukov, 2017-11-01:
Hello,

I am running 32bit gentoo linux 4.14.0-rc5-v7+ on rpi 3B which is connected to a Samsung TV via HDMI.

firmware files from master branch at https://github.com/RPi-Distro/firmware-nonfree, copied those into /boot:
start_cd.elf, start.elf, bootcode.bin, fixup_cd.dat, fixup.dat

/boot/config.txt related settings:
#kernel vc4+cec
dtoverlay=vc4-kms-v3d,cma-128
#mask_gpu_interrupt1=0x100
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=26370
#end kernel vc4+cec

#cec detection
hdmi_force_hotplug=1
#end cec detection

I have tried without the gpu masks and with different combinations, but had no luck.

libdrm mesa xorg-drivers are all compiled with vc4 video card support in userspace.

libcec vesion: dev-libs/libcec-4.0.2:0

If I disable the dtoverlay setting in config.txt and make use of the fbdev video driver, then, via cec and pulse8_cec kernel modules I can get cec-client to identify my TV and capture remote control key-pressed events. Same applies for kodi, but kodi is extremely slow with this. So the HDMI cable and TV work fine with CEC.

However, if I use the vc4-kms-v3d setting and the vc4 kernel driver the TV is not identified properly and remote control key-presses are not recognized.

Anyone else experiencing this same behaviour? Thought that according to:
https://lists.freedesktop.org/archives/dri-devel/2017-July/146710.html
and https://patchwork.kernel.org/patch/9834523/

CEC should be functional in 4.13+ kernels.

Not sure if this is a configuration issue or a bug. Any help is greatly appreciated.

Thank you,
-Nikolay

nkichukov, 2017-11-02:
I have also removed DRM_VC4_HDMI_CEC from the kernel image, with various gpu interrupt masks combinations enabled, but still no luck. The goal was to have vc4-kms-v3d enabled and vc4 kernel driver in use, but reroute cec through the legacy cec channels, if at all possible.

Cheers,
-Nik

anholt, 2017-11-08:
If you're using the open source driver, you also need to use the open source CEC support -- CEC is in the HDMI module, so you can't mix-and-match.

I'm not sure if libcec supports the kernel's CEC implementation yet -- for a long time it didn't.  For my testing I used cec-ctl from v4l-utils.

nkichukov, 2017-11-09:
Understood.
Asked with kodi for direct support without libcec - not a chance.
Now asking in #pulse-eight and still awaiting response...

Thank you for clarification, closing this.
-N

closed 2017-11-09 by nkichukov