anholt/linux #119: vc4: Failed to release power domain in 4.14rc7

lategoodbye, 2017-11-07:
On my Raspberry Pi 1 B i always see the following error in dmesg after boot:

[ 147.081470] [drm:vc4_vec_encoder_disable] *ERROR* Failed to release power domain: -1

I cannot see any negative side effects, but never seen this issue before.

config: bcm2835_defconfig
Kernel version: 4.14rc7

anholt, 2017-11-08:
Hmm.  I'm surprised we've got a `vc4_vec_encoder_disable()` being called, and I wonder what the backtrace is.

lategoodbye, 2017-11-09:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-rc7+ (stefan@Latitude-E4310) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #5 Thu Nov 9 08:06:33 CET 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 32 MiB at 0x0a000000
[    0.000000] On node 0 totalpages: 49152
[    0.000000] free_area_init_node: node 0, pgdat c0c72014, node_mem_map c9e6a000
[    0.000000]   Normal zone: 384 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 49152 pages, LIFO batch:15
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] random: fast init done
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 48768
[    0.000000] Kernel command line: console=tty0 console=ttyAMA0 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait noinitrd
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 149356K/196608K available (7168K kernel code, 575K rwdata, 2196K rodata, 1024K init, 687K bss, 14484K reserved, 32768K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xcc800000 - 0xff800000   ( 816 MB)
                   lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc0800000   (8160 kB)
                     .init : 0xc0b00000 - 0xc0c00000   (1024 kB)
                     .data : 0xc0c00000 - 0xc0c8fd68   ( 576 kB)
                      .bss : 0xc0c9631c - 0xc0d4218c   ( 688 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25445 entries in 75 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000034] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000081] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000184] bcm2835: system timer (irq = 27)
[    0.000882] Console: colour dummy device 80x30
[    0.001536] console [tty0] enabled
[    0.001604] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.050379] pid_max: default: 32768 minimum: 301
[    0.050887] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.050947] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.052183] CPU: Testing write buffer coherency: ok
[    0.053513] Setting up static identity map for 0x100000 - 0x100054
[    0.058280] devtmpfs: initialized
[    0.065906] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.066424] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.066497] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.072271] pinctrl core: initialized pinctrl subsystem
[    0.073105] NET: Registered protocol family 16
[    0.075171] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.079741] No ATAGs?
[    0.079783] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.079876] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.080256] Serial: AMBA PL011 UART driver
[    0.083808] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_gpio14, deferring probe
[    0.103610] SCSI subsystem initialized
[    0.104159] usbcore: registered new interface driver usbfs
[    0.104338] usbcore: registered new interface driver hub
[    0.104568] usbcore: registered new device driver usb
[    0.105549] Advanced Linux Sound Architecture Driver Initialized.
[    0.108101] clocksource: Switched to clocksource timer
[    0.201640] simple-framebuffer e330000.framebuffer: framebuffer at 0xe330000, 0x8ca000 bytes, mapped to 0xcd000000
[    0.201754] simple-framebuffer e330000.framebuffer: format=a8r8g8b8, mode=1920x1200x32, linelength=7680
[    0.307407] Console: switching to colour frame buffer device 240x75
[    0.412182] simple-framebuffer e330000.framebuffer: fb0: simplefb registered!
[    0.434276] NET: Registered protocol family 2
[... 179 lines cut ...]
[   56.195857] [<c010c78c>] (show_stack) from [<c071c48c>] (dump_stack+0x20/0x28)
[   56.195902] [<c071c48c>] (dump_stack) from [<c011e758>] (__warn+0xe4/0x110)
[   56.195926] [<c011e758>] (__warn) from [<c011e854>] (warn_slowpath_null+0x30/0x38)
[   56.195957] [<c011e854>] (warn_slowpath_null) from [<c043c9a0>] (vc4_vec_encoder_disable+0x80/0xa8)
[   56.195997] [<c043c9a0>] (vc4_vec_encoder_disable) from [<c0404100>] (drm_atomic_helper_commit_modeset_disables+0x118/0x3cc)
[   56.196041] [<c0404100>] (drm_atomic_helper_commit_modeset_disables) from [<c0438764>] (vc4_atomic_complete_commit+0x44/0xc8)
[... cut ...]
[   56.196344] [<c03fdaa0>] (output_poll_execute) from [<c0133d1c>] (process_one_work+0x204/0x380)
[   56.196364] [<c0133d1c>] (process_one_work) from [<c0134c44>] (worker_thread+0x2e0/0x450)
[   56.196404] [<c0134c44>] (worker_thread) from [<c01399e8>] (kthread+0x140/0x15c)
[   56.196438] [<c01399e8>] (kthread) from [<c0107e70>] (ret_from_fork+0x14/0x24)
[   56.196448] ---[ end trace 3e8f278286937bd6 ]---
[   56.196471] [drm:vc4_vec_encoder_disable] *ERROR* Failed to release power domain: -1
[   57.310292] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SS
```

lategoodbye, 2017-11-12:
It seems to me rpm_check_suspend_allowed() returns with -EPERM in this case.

anholt, 2017-11-14:
The vec encoder should be disabled by default.  Do you have something else on top that would be enabling vec?

anholt, 2017-11-14:
I was totally thinking of the wrong encoder, and VEC is just fine.  I've tried to turn VEC on and off with:

```
xrandr --output Composite-1 --mode 720x480
xrandr --output Composite-1 --off
```

and not reproduced on rc5.

lategoodbye, 2017-11-14:
The issue seems to be triggered by using my HDMI switch or unplug and reconnecting.

Additionally i tried to narrow down the issue for this specific version 4.14rc7:
RPI 1 B + bcm2835_defconfig => issue reproducible
RPI 2 B + multi_v7_defconfig => issue reproducible
RPI 2 B + bcm2835_defconfig => issue reproducible

anholt, 2017-11-14:
Yeah, I tried rc7 this morning on my normal config, and it worked. I'm guessing there's some piece of RPM we need to enable to make bcm2835_defconfig work.

lategoodbye, 2017-11-16:
I agree that we need to figure out the responsible config option, but i would expect vc4 works with current defconfig, too.

closed 2019-12-05 by lategoodbye