anholt/linux #120: vc4: Memory leaks in linux-next-20171110

lategoodbye, 2017-11-12:
If i boot RPi 3 (ARM64, defconfig) with insufficient CMA memory, i'll get a lot of memory leaks:

```
[    7.972104] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
[    7.980095] vc4_hdmi 3f902000.hdmi: ASoC: no DMI vendor name!
[    8.000121] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[    8.011158] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[    8.018605] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    8.026034] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.034798] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.txt failed with error -2
[    8.046524] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.055081] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[    8.063117] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[    8.070395] vc4_v3d 3fc00000.v3d: Failed to allocate memory for tile binning: -12. You may need to enable CMA or give it more memory.
[    8.082893] vc4-drm soc:gpu: failed to bind 3fc00000.v3d (ops vc4_v3d_ops [vc4]): -12
[    8.088377] vc4-drm soc:gpu: master bind failed: -12
[    8.088417] vc4-drm: probe of soc:gpu failed with error -12
...
[  678.403983] kmemleak: 36 new suspected memory leaks (see /sys/kernel/debug/kmemleak)
```

Here is the output of kmemleak (i've cut it down because the last 2 entries seems to repeat):

```
unreferenced object 0xffff800036c55918 (size 576):
  comm "systemd-udevd", pid 1951, jiffies 4294894285 (age 754.836s)
  hex dump (first 32 bytes):
    00 00 39 00 00 00 00 00 98 bd da 36 00 80 ff ff  ..9........6....
    78 b2 25 37 00 80 ff ff 30 59 c5 36 00 80 ff ff  x.%7....0Y.6....
  backtrace:
    [<ffff0000081f08d8>] kmem_cache_alloc+0x1d8/0x220
    [<ffff0000089bb1c0>] radix_tree_node_alloc.constprop.12+0x50/0xe8
    [<ffff0000089bcae0>] idr_get_free_cmn+0x2d8/0x320
    [<ffff0000089b60e8>] idr_alloc_cmn+0x58/0xd8
    [<ffff000000b42aac>] __drm_mode_object_add+0x5c/0xa8 [drm]
    [<ffff000000b42b0c>] drm_mode_object_add+0x14/0x20 [drm]
    [<ffff000000b432dc>] drm_property_create+0xb4/0x168 [drm]
    [<ffff000000b419a0>] drm_connector_create_standard_properties+0x20/0xf0 [drm]
    [<ffff000000b46598>] drm_mode_config_init+0xe8/0x488 [drm]
    [<ffff000000b92328>] vc4_drm_bind+0x88/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
    [<ffff0000085b1fa4>] __driver_attach+0xac/0xb0
unreferenced object 0xffff80003524c780 (size 128):
  comm "systemd-udevd", pid 1951, jiffies 4294894285 (age 754.840s)
  hex dump (first 32 bytes):
    70 69 78 65 6c 00 ff ff 00 d0 69 0d 00 00 ff ff  pixel.....i.....
    02 00 00 00 00 00 00 00 98 c7 24 35 00 80 ff ff  ..........$5....
  backtrace:
    [<ffff0000081f3880>] __kmalloc_track_caller+0x210/0x298
    [<ffff0000081a9c00>] kstrdup+0x38/0x68
    [<ffff0000081a9c64>] kstrdup_const+0x34/0x40
    [<ffff0000084df0b0>] __clk_create_clk.part.22+0x50/0xb0
    [<ffff0000084dfcb4>] __of_clk_get_from_provider+0x114/0x168
    [<ffff0000084db8f4>] __of_clk_get_by_name+0xfc/0x138
    [<ffff0000084db984>] clk_get+0x2c/0x78
    [<ffff0000084dae58>] devm_clk_get+0x48/0xa0
    [<ffff000000b99290>] vc4_hdmi_bind+0xa0/0x5f0 [vc4]
    [<ffff0000085abfdc>] component_bind_all+0xfc/0x258
    [<ffff000000b9233c>] vc4_drm_bind+0x9c/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
unreferenced object 0xffff800037fff000 (size 128):
  comm "systemd-udevd", pid 1951, jiffies 4294894302 (age 754.772s)
  hex dump (first 32 bytes):
    70 6c 61 6e 65 2d 35 00 00 d0 cf 09 00 00 ff ff  plane-5.........
    02 00 00 00 00 00 00 00 18 f0 ff 37 00 80 ff ff  ...........7....
  backtrace:
    [<ffff0000081f3880>] __kmalloc_track_caller+0x210/0x298
    [<ffff0000083c6be8>] kvasprintf+0x68/0xd0
    [<ffff0000083c6d58>] kasprintf+0x48/0x50
    [<ffff000000b447ac>] drm_universal_plane_init+0x354/0x4d8 [drm]
    [<ffff000000b9cd58>] vc4_plane_init+0xb0/0xf0 [vc4]
    [<ffff000000b915d8>] vc4_crtc_bind+0xe0/0x380 [vc4]
    [<ffff0000085abfdc>] component_bind_all+0xfc/0x258
    [<ffff000000b9233c>] vc4_drm_bind+0x9c/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
    [<ffff0000085b1fa4>] __driver_attach+0xac/0xb0
    [<ffff0000085affdc>] bus_for_each_dev+0x64/0xa0
    [<ffff0000085b1700>] driver_attach+0x20/0x28
unreferenced object 0xffff800037fff600 (size 128):
  comm "systemd-udevd", pid 1951, jiffies 4294894302 (age 754.780s)
  hex dump (first 32 bytes):
    58 52 32 34 41 52 32 34 41 42 32 34 58 42 32 34  XR24AR24AB24XB24
    52 47 31 36 42 47 31 36 41 52 31 35 58 52 31 35  RG16BG16AR15XR15
  backtrace:
    [<ffff0000081f0438>] __kmalloc+0x210/0x298
    [<ffff000000b444fc>] drm_universal_plane_init+0xa4/0x4d8 [drm]
    [<ffff000000b9cd58>] vc4_plane_init+0xb0/0xf0 [vc4]
    [<ffff000000b915d8>] vc4_crtc_bind+0xe0/0x380 [vc4]
    [<ffff0000085abfdc>] component_bind_all+0xfc/0x258
    [<ffff000000b9233c>] vc4_drm_bind+0x9c/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
    [<ffff0000085b1fa4>] __driver_attach+0xac/0xb0
    [<ffff0000085affdc>] bus_for_each_dev+0x64/0xa0
    [<ffff0000085b1700>] driver_attach+0x20/0x28
    [<ffff0000085b11a0>] bus_add_driver+0x108/0x228
    [<ffff0000085b2940>] driver_register+0x60/0xf8
unreferenced object 0xffff800037fffa80 (size 128):
  comm "systemd-udevd", pid 1951, jiffies 4294894302 (age 754.780s)
  hex dump (first 32 bytes):
    70 6c 61 6e 65 2d 38 00 00 50 d6 09 00 00 ff ff  plane-8..P......
    02 00 00 00 00 00 00 00 98 fa ff 37 00 80 ff ff  ...........7....
  backtrace:
    [<ffff0000081f3880>] __kmalloc_track_caller+0x210/0x298
    [<ffff0000083c6be8>] kvasprintf+0x68/0xd0
    [<ffff0000083c6d58>] kasprintf+0x48/0x50
    [<ffff000000b447ac>] drm_universal_plane_init+0x354/0x4d8 [drm]
    [<ffff000000b9cd58>] vc4_plane_init+0xb0/0xf0 [vc4]
    [<ffff000000b915d8>] vc4_crtc_bind+0xe0/0x380 [vc4]
    [<ffff0000085abfdc>] component_bind_all+0xfc/0x258
    [<ffff000000b9233c>] vc4_drm_bind+0x9c/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
    [<ffff0000085b1fa4>] __driver_attach+0xac/0xb0
    [<ffff0000085affdc>] bus_for_each_dev+0x64/0xa0
    [<ffff0000085b1700>] driver_attach+0x20/0x28
unreferenced object 0xffff800037e03b80 (size 128):
  comm "systemd-udevd", pid 1951, jiffies 4294894307 (age 754.764s)
  hex dump (first 32 bytes):
    58 52 32 34 41 52 32 34 41 42 32 34 58 42 32 34  XR24AR24AB24XB24
    52 47 31 36 42 47 31 36 41 52 31 35 58 52 31 35  RG16BG16AR15XR15
  backtrace:
    [<ffff0000081f0438>] __kmalloc+0x210/0x298
    [<ffff000000b444fc>] drm_universal_plane_init+0xa4/0x4d8 [drm]
    [<ffff000000b9cd58>] vc4_plane_init+0xb0/0xf0 [vc4]
    [<ffff000000b915d8>] vc4_crtc_bind+0xe0/0x380 [vc4]
    [<ffff0000085abfdc>] component_bind_all+0xfc/0x258
    [<ffff000000b9233c>] vc4_drm_bind+0x9c/0x168 [vc4]
    [<ffff0000085abc3c>] try_to_bring_up_master+0x144/0x1a8
    [<ffff0000085abe9c>] component_master_add_with_match+0x9c/0xe0
    [<ffff000000b924bc>] vc4_platform_drm_probe+0xb4/0xf0 [vc4]
    [<ffff0000085b3c30>] platform_drv_probe+0x58/0xc0
    [<ffff0000085b1e14>] driver_probe_device+0x224/0x308
    [<ffff0000085b1fa4>] __driver_attach+0xac/0xb0
    [<ffff0000085affdc>] bus_for_each_dev+0x64/0xa0
    [<ffff0000085b1700>] driver_attach+0x20/0x28
    [<ffff0000085b11a0>] bus_add_driver+0x108/0x228
    [<ffff0000085b2940>] driver_register+0x60/0xf8
...
```

lategoodbye, 2017-11-14:
In a naive attempt i tried the following patch, which seems to fix the leaks:

```
diff --git a/drivers/gpu/drm/vc4/vc4_drv.c b/drivers/gpu/drm/vc4/vc4_drv.c
index e3c2972..07cd177 100644
--- a/drivers/gpu/drm/vc4/vc4_drv.c
+++ b/drivers/gpu/drm/vc4/vc4_drv.c
@@ -291,6 +291,7 @@ static int vc4_drm_bind(struct device *dev)
        component_unbind_all(dev, drm);
 gem_destroy:
        vc4_gem_destroy(drm);
+       drm_mode_config_cleanup(drm);
        vc4_bo_cache_destroy(drm);
 dev_unref:
        drm_dev_unref(drm);
```

Unfortunately this also triggers the following warnings:

lategoodbye, 2017-11-17:
I reduced the CMA to 16 MB on Raspberry Pi Zero W with Linux 4.14rc7 and get the same memleak. So it seems to be older.

nkichukov, 2017-11-29:
Hi, I have created a similar issue report: https://github.com/anholt/linux/issues/122
I am not sure if those two are the same issue or not as mine happens with a different CMA size + the trace looks different.

b/drivers/gpu/drm/vc4/vc4_drv.c - did you attempt to write a self-fix for the mem-leak or was this provided by the author to test?

Thank you,
-Nik

lategoodbye, 2017-11-29:
I looked at other drm drivers like sun4i_drv.c which do similiar things and found out that they call drm_mode_config_cleanup() in the error path of the probe function. Maybe i need to change the order.

closed 2019-12-05 by lategoodbye