anholt/linux #125: Crash in __irq_startup (chip.c)

fluffysheap, 2017-12-26:
I have been trying to use recent KMS drivers with Raspberry Pi since the KMS driver included with official 4.14 kernel [doesn't work with the DSI display](https://github.com/raspberrypi/firmware/issues/909).  To that end, I built a kernel from the latest torvalds repository on kernel.org, which was 4.15-rc5.

I installed the kernel, modules, and generated device trees into an otherwise unmodified Raspbian Stretch image with full userspace updates and firmware 12-1-2017, and booted on a version 1.1 Pi2 with the official DSI display.  Upon booting the backlight was on, but nothing appeared on the screen.  I noticed that the usual vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4]) line is not present.  Instead I got a stack trace.  I have also attached my kernel config file, config.txt and cmdline.txt.
[vc4-configs.txt](https://github.com/anholt/linux/files/1587140/vc4-configs.txt)
[kconfig-based-on-49.gz](https://github.com/anholt/linux/files/1587142/kconfig-based-on-49.gz)
[vc4-trace.txt](https://github.com/anholt/linux/files/1587925/vc4-trace.txt)

Now, because this was a kernel configuration I made myself based on RPi's 4.9 kernel config, I tried again using multi_v7_defconfig.  I installed this into my Gentoo userspace, which works great using the 4.9 kernel with fkms and also includes the 12-1-2017 firmware.  I got the same stack trace.  I attached the config files (except the kernel config since it's a straight multi_v7_defconfig), dmesg and Xorg log.
[dmesg.log](https://github.com/anholt/linux/files/1587954/dmesg.log)
[mv7-configs.txt](https://github.com/anholt/linux/files/1587955/mv7-configs.txt)
[Xorg.0.log](https://github.com/anholt/linux/files/1587956/Xorg.0.log)

In config.txt I've tried both with and without ignore_lcd=1.

With the v7_defconfig, the console works at boot, but somewhere along the way - probably at the time of the stack trace - stops working and instead there's a white screen.

The (hopefully) relevant bit of the kernel log, excerpted from dmesg.log, follows:
[   18.843924] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
[   18.843956] vc4_hdmi 3f902000.hdmi: ASoC: no DMI vendor name!
[   18.852887] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   18.853178] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[   18.853295] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   18.853605] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.853891] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.854157] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.890950] ------------[ cut here ]------------
[   18.890983] WARNING: CPU: 3 PID: 723 at kernel/irq/chip.c:244 __irq_startup+0x94/0xa8
[   18.890988] Modules linked in: vc4(+) snd_soc_core ac97_bus snd_pcm_dmaengine snd_pcm snd_timer snd soundcore cec
[   18.891036] CPU: 3 PID: 723 Comm: udevd Not tainted 4.15.0-rc5 #1
[   18.891041] Hardware name: BCM2835
[   18.891075] [<c0311b34>] (unwind_backtrace) from [<c030ca6c>] (show_stack+0x10/0x14)
[   18.891093] [<c030ca6c>] (show_stack) from [<c0d2beb8>] (dump_stack+0x8c/0xa0)
[   18.891113] [<c0d2beb8>] (dump_stack) from [<c0342014>] (__warn+0xe0/0xf8)
[   18.891131] [<c0342014>] (__warn) from [<c03420fc>] (warn_slowpath_null+0x40/0x48)
[   18.891148] [<c03420fc>] (warn_slowpath_null) from [<c0392a8c>] (__irq_startup+0x94/0xa8)
[   18.891166] [<c0392a8c>] (__irq_startup) from [<c0392aec>] (irq_startup+0x4c/0x130)
[   18.891182] [<c0392aec>] (irq_startup) from [<c03907c4>] (enable_irq+0x34/0x6c)
[   18.891327] [<c03907c4>] (enable_irq) from [<bf0a4ab4>] (vc4_irq_postinstall+0x14/0x34 [vc4])
[   18.891432] [<bf0a4ab4>] (vc4_irq_postinstall [vc4]) from [<c082b05c>] (drm_irq_install+0xd0/0x13c)
[   18.891523] [<c082b05c>] (drm_irq_install) from [<bf0a7938>] (vc4_v3d_bind+0x130/0x224 [vc4])
[   18.891614] [<bf0a7938>] (vc4_v3d_bind [vc4]) from [<c08bd58c>] (component_bind_all+0xfc/0x22c)
[   18.891701] [<c08bd58c>] (component_bind_all) from [<bf09b0f4>] (vc4_drm_bind+0x94/0x13c [vc4])
[   18.891788] [<bf09b0f4>] (vc4_drm_bind [vc4]) from [<c08bd974>] (try_to_bring_up_master+0x140/0x17c)
[   18.891806] [<c08bd974>] (try_to_bring_up_master) from [<c08bdb98>] (component_master_add_with_match+0xa4/0xd4)
[   18.891891] [<c08bdb98>] (component_master_add_with_match) from [<bf09b244>] (vc4_platform_drm_probe+0xa8/0xb8 [vc4])
[   18.891978] [<bf09b244>] (vc4_platform_drm_probe [vc4]) from [<c08c47cc>] (platform_drv_probe+0x50/0xac)
[   18.891994] [<c08c47cc>] (platform_drv_probe) from [<c08c2d50>] (driver_probe_device+0x254/0x32c)
[   18.892009] [<c08c2d50>] (driver_probe_device) from [<c08c2ecc>] (__driver_attach+0xa4/0xa8)
[   18.892026] [<c08c2ecc>] (__driver_attach) from [<c08c127c>] (bus_for_each_dev+0x4c/0x9c)
[   18.892041] [<c08c127c>] (bus_for_each_dev) from [<c08c2290>] (bus_add_driver+0x18c/0x210)
[   18.892055] [<c08c2290>] (bus_add_driver) from [<c08c39cc>] (driver_register+0x78/0xf4)
[   18.892070] [<c08c39cc>] (driver_register) from [<c03021bc>] (do_one_initcall+0x44/0x168)
[   18.892090] [<c03021bc>] (do_one_initcall) from [<c03c57e8>] (do_init_module+0x64/0x1f4)
[   18.892108] [<c03c57e8>] (do_init_module) from [<c03c49c0>] (load_module+0x1d9c/0x23e8)
[   18.892126] [<c03c49c0>] (load_module) from [<c03c5214>] (SyS_finit_module+0x9c/0xb4)
[   18.892143] [<c03c5214>] (SyS_finit_module) from [<c03081e0>] (ret_fast_syscall+0x0/0x54)
[   18.892151] ---[ end trace dbeb6479acc03010 ]---
[   18.892270] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   18.892287] checking generic (3eb3f000 bb800) vs hw (0 ffffffff)
[   18.892292] fb: switching to vc4drmfb from simple
[   18.892453] Console: switching to colour dummy device 80x30
[   18.900356] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   18.900372] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   18.900377] [drm] Driver supports precise vblank timestamp query.
[   18.951311] Console: switching to colour frame buffer device 90x30
[   18.966536] vc4-drm soc:gpu: fb0:  frame buffer device

stschake, 2017-12-26:
The warning isn't stopping you from DSI, it's just that - a warning. There is a thread on dri-devel where we discussed it:

https://lists.freedesktop.org/archives/dri-devel/2017-December/thread.html#159917

I have a plan for a fix but I lack the hardware right now that could reproduce the original race & crash.

The problem here is simply that the upstream device tree isn't configured to use the DSI display. You'll need to adapt this patch from the raspberry tree for it to work:

https://github.com/raspberrypi/linux/commit/b7fe2e5fd1132fc92878edd2d475606e6fe85d9f#diff-bd97b00ee6a11577d77425f50b39a39a

The integration of the display into the upstream DT is currently blocked on this:

https://lists.freedesktop.org/archives/dri-devel/2017-November/157688.html

lategoodbye, 2017-12-27:
Regarding to the upstream DT there is already a patch from Eric, which needs less adaption:
https://patchwork.kernel.org/patch/9790281/

@fluffysheap In case of RPI2 you will need to connect I2C of the panel to pins 28, 29.

@stschake This warning is quiet annoying and more users will soon complain about it because it stays in 4.15. What do you need to get this fixed?

stschake, 2017-12-27:
@lategoodbye the problem is reproducing the original kernel OOPS in overflow_mem_work to verify a patch. I don't have access right now to the one device that did it regularly in the past. In theory it is all the same hardware but in practice there are a few of them that kept hitting the right timing to the point of being stuck in boot loops.

fluffysheap, 2017-12-29:
@lategoodbye Thanks for your help.  The first patch I was able to port over, but it was missing the port sections.  With the port sections, it didn't appear to work at first (-517 error), but then with a followup manual module load, it started working.  Upside down and no touchscreen, but I might be on the way.  Having to try twice to load the driver is not great, hopefully the defer thing can get sorted out, but at least it's working now.

fluffysheap, 2017-12-29:
I guess I should leave the issue open since the warning, at least, is real, even though my immediate problem is fixed.

anholt, 2018-01-04:
Fixed in:
```
commit a23a85a37dee8beccf0bb6a7dffccddf404cb97f
Author: Stefan Schake <stschake@gmail.com>
Date:   Fri Dec 29 17:05:43 2017 +0100

    drm/vc4: Move IRQ enable to PM path
```

closed 2018-01-04 by anholt