anholt/linux #128: display turns off after sporadic 'flip_done timeout error'

nkichukov, 2018-02-11:
Hi folks,

The reported issue is sporadic and not reproducible on demand. As per the log below, it has happened 3 times in 21 days. All 3 times, the display went blank and got powered off until the Xorg session was restarted.

Details follow below:
kernel: 4.15.0-rc7-v7+ + patch *1.) which fixed memory leak issue addressed at #123 
compiler: gcc-6.4
raspberry pi running in 32bit mode
CMA=256MB
OS: Gentoo Linux

Error printed in the kernel log:
[1042617.315980] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:53:crtc-2] flip_done timed out
[1042842.595966] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:52:plane-20] flip_done timed out
[1042959.331986] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:53:crtc-2] flip_done timed out
[1043354.595985] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:52:plane-20] flip_done timed out
[1761553.894440] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:53:crtc-2] flip_done timed out
[1762164.198463] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:52:plane-20] flip_done timed out

Workaround:
terminate X11 session and restart it.

Thank you,
-Nikolay

*1.) https://lists.freedesktop.org/archives/dri-devel/2018-January/161625.html

lategoodbye, 2018-02-12:
Please follow this thread:
https://lists.freedesktop.org/archives/dri-devel/2018-February/165428.html

nkichukov, 2018-04-03:
Thanks for the pointers Stefan,
I have attempted rpi-4.16.0-rc7 kernel which does include the clk-patches from https://patchwork.kernel.org/patch/10207161/raw/ and drm/atomic fixes from https://patchwork.kernel.org/patch/10169179/raw/ however, I am still getting the same sporadic results:

`[ 5433.053889] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:68:crtc-2] flip_done timed out
[ 5551.325851] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:67:plane-20] flip_done timed out
`
Any other pointers/patches?
-Nik

nkichukov, 2018-04-03:
Reproduced with drm.debug=0x3f. A lot of repeated messages, not sure if this will help. Leaving the kern.log.gz file available for download(10MB), uncompressed 142MB.

http://www.oldum.net/~nikolay/kern.log.gz

First occurrence:
Apr  3 23:34:23 grpi kernel: [  906.205803] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:68:crtc-2] flip_done timed out
Second:
Apr  3 23:38:18 grpi kernel: [ 1141.213846] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:67:plane-20] flip_done timed out

Cheers,
-N

nkichukov, 2018-04-14:
Upgraded to rpi-4.16.1 which seems to have received multiple vc4 patches and I am unable to reproduce the flip_done timeout for the last 3 days.

nkichukov, 2018-04-18:
day 6, issue reoccurred:
```
[516937.439594] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:68:crtc-2] flip_done timed out
[517547.743601] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:67:plane-20] flip_done timed out

```

nkichukov, 2018-05-28:
This continues to occur, so sporadic as it was...
Changes attempted:
Running 4.16.10 kernel now, noticed loads of vc4 commits made it there, to no avail for this bug unfortunately.
Running latest version of libdrm, from https://anongit.freedesktop.org/git/mesa/drm.git, no change.
Version of mesa: mesa-18.1.0, tried with experimental from git, same results.
Xorg server: xorg-server-1.19.5-r2
compiler upgraded to: gcc-7.3.0-r3 and all of the above recompiled and with -O2, same results.

``[74697.182124] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:68:crtc-2] flip_done timed out
[74814.686141] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:67:plane-20] flip_done timed out
[124903.134330] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:68:crtc-2] flip_done timed out
[125513.438329] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:67:plane-20] flip_done timed out
``

Any pointers? Where to go next?
Cheers,
-N

nkichukov, 2018-06-21:
This is no longer happening, so I am closing the case. It seems like it was not due to the driver/kernel side, but userland, as the few changes I've made about 20 days ago were to upgrade xorg-server and xorg-drivers to version 1.20.0 and also disable some of the enabled but unused --configure switches for mesa when compiling it. This got the system back to a stable state and the flip_done timeout is no longer present, at least not in the last 20 days...

Cheers,
-N

closed 2018-06-21 by nkichukov

lategoodbye, 2019-01-25:
Unfortunately this "flip_done timed out" still occurs e.g. next-20190125. 

Unfortunately i don't know the used display on the kernelci machine (Raspberry Pi 3 B, arm64/defconfig):
https://kernelci.org/boot/id/5c4ab5b359b514f4f39ad006/

```
07:07:28.667279  [   26.147366] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:139:crtc-2] flip_done timed out
07:07:28.668117  [   26.147369] Bluetooth: hci0: Failed to set baudrate
07:07:28.702757  [   26.147527] Console: switching to colour frame buffer device 90x30
07:07:28.705750  [   26.259404] Bluetooth: hci0: BCM: chip id 94
07:07:28.706766  [   26.261556] Bluetooth: hci0: BCM: features 0x2e
07:07:28.707637  [   26.284543] Bluetooth: hci0: BCM43430A1
07:07:28.708475  [   26.284556] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
07:07:28.709290  [   26.284658] bluetooth hci0: Direct firmware load for brcm/BCM43430A1.hcd failed with error -2
07:07:28.710129  [   26.284668] Bluetooth: hci0: BCM: Patch brcm/BCM43430A1.hcd not found
07:07:28.749202  [   36.387352] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:139:crtc-2] flip_done timed out
07:07:28.751899  [   45.569912] NET: Registered protocol family 10
07:07:28.752647  [   45.571503] Segment Routing with IPv6
07:07:28.753346  [   46.627353] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:46:Composite-1] flip_done timed out
07:07:28.754030  [   56.867345] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:135:plane-20] flip_done timed out
07:07:28.756490  [   57.030633] vc4-drm soc:gpu: fb0: DRM emulated frame buffer device
```