anholt/linux #130: HifiBerry+ DAC and vc4hdmi: different default ALSA card bewteen reboots

vanfanel, 2018-04-04:
Hi there, Eric

I updated my system to the latest kernel (4.14.31-v7+) and now I get different default sound cards between boots.
Sometimes aplay -L will report:
default:CARD=sndrpihifiberry
and sometimes it will report:
default:CARD=vc4hdmi

When the vc4hdmi (which I don't currently use) is the default card, sound won't work and alsamixer reports it having no controls.

Do you have an idea on what could be going on?
Any way to disable vc4hdmi so it's never used on the Pi where I use HifiBerry+?

anholt, 2018-04-09:
My guess would be that baseline ALSA is just defaulting to whatever module probes first, and module probe order in Linux is arbitrary.  You should probably be using some software like pulseaudio that implements policy on top of ALSA, or just force the issue using alsa.conf snippets somehow.

vanfanel, 2018-04-10:
@anholt : But I am not using the vc4-hdmi audio. What modules does it use specifically? I don't see a specific vc4-hdmi audio module on lsmod. If I knew what module does vc4-hdmi audio use, I could simply prevent it from being loaded in this machine.

stschake, 2018-04-10:
There is no specific module for it - if you use the VC4 DRM driver, you will automatically get the HDMI audio.

vanfanel, 2018-04-10:
@stschake : I see. Then I don't see the way around it by not loading a module :(
@anholt : No way I will be even using PulseAudio. As for ALSA, one can easily chose the default sound card using something like:

```
pcm.!default {
	type hw
	card 0
}

ctl.!default {
	type hw           
	card 0
}

```
In asound.conf. The problem is that the card number assigned to the hifiberry and the vc4hdmi changes bewteen boots. Is there a way to tell default by name instead of number?

vanfanel, 2019-02-08:
@anholt 
I still get problems with arbitrary card ordering.
It should be possible to tell Linux probing to NEVER let vc4-hdmi to be the first card, by passing an option. So, assuming the module responsible for vc4-hdmi audio card is called "vc4" (since such a module exists when I do `lsmod`) I created /etc/modprobe.d/alsa-base.conf with this content:

`options vc4 index=-2`

I got the information for this here:
https://wiki.archlinux.org/index.php/Advanced_Linux_Sound_Architecture

However, from time to time, vc4-hdmi is STILL alsa card 0, causing the alsa-restore.service to fail on system boot.
So, please, how can I "convince"  the kernel to never make vc4-hdmi the first ALSA card, once and for all?

pelwell, 2019-02-08:
Alternatively, is there a way to disable the HDMI audio device? If there's a suitable DT property value then I can add a parameter to the vc4-kms-v3d overlay.

6by9, 2019-02-08:
It looks like nuking the dmas property from the hdmi DT node will do it, whilst leaving the rest of the vc4 driver to come up. Not the cleanest, but might be worth a quick try.

https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/vc4/vc4_hdmi.c#L1075

pelwell, 2019-02-08:
If only an overlay could delete a property. <sad face>