anholt/linux #131: DSI 1 - How many Lanes?

mbt28, 2018-04-17:
Hi,

In the vc4_dsi.c file you have been mentioning that DSI1 has 4 lanes but when I check the Rpi 3 b+ schematics I only see 1 clock lane and 2 data lanes. The other 2 lanes arent exposed to the connector or they are not just visible on the schematics?

 * BCM2835 contains two DSI modules, DSI0 and DSI1.  DSI0 is a
 * single-lane DSI controller, while DSI1 is a more modern 4-lane DSI
 * controller.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_3bplus_1p0_reduced.pdf

Thanks

anholt, 2018-04-17:
The normal Pi boards look like they only connect up two lanes, you're right.  However, there are 4 for DSI1, and the CM3 boards do bring out all 4.  I've updated the wiki page with a bit more about it.

mbt28, 2018-04-18:
Hi Eric,

Then with 2 lanes we can only drive HD resolution screens. I cant find anything on internet the specs of DSI on pi however it seems max speed is 1 Gbps Per Lane. But I am not sure I see some implementations of 1.5 Gbps per lane as well. 

If it is 1 gbps max then according to my calculations we can only make around 60-fps 1280 × 720 Resolution at 24-bpp.

Another question what is the logic level of DSI lanes? As I see on the internet it has to be 1.2V, I cant be sure about it.

anholt, 2018-04-18:
Yeah, the DSI PHY (bit) clocks can go up to 1Ghz.  And it seems to be 1.2V.

mbt28, 2018-04-18:
Hi,

I think my calculations are wrong, theoretically pi should be able to drive 1080p screen with 2 lanes. 

If anybody interested, here is the calculation for 1920x1080p@60Hz, RAW10, 2-lane:
Total Horizontal Samples = 2200, Total Vertical Lines = 1125
Pixel Clock Frequency = 2200 x 1125 x 60 = 148.5 MHz
Bandwidth (Total Data Rate) = 148.5 MHz * 10-bit = 1.485 Gbps
Line Rate (Data Rate per Lane) = 1.485 Gbps / 2-lane = 742.5 Mbps
MIPI Bit Clock Frequency = 742.5 / 2 = 371.25 MHz

http://www.latticesemi.com/-/media/LatticeSemi/Documents/UserManuals/MQ/UG110.ashx?document_id=51382

closed 2018-04-18 by mbt28

anholt, 2018-04-18:
I don't think we can go under rgb565 depth, though.

mbt28, 2018-04-18:
> I don't think we can go under rgb565 depth, though.

ohh my mistake somehow I saw this 10 like 10 bits per pixel. Even if we do rgb565 seems unfortunately it is beyonds limits of 2 lanes.