anholt/linux #133: transfer interrupt wait timeout on new DSI Panel

jwaani2, 2018-05-03:
Hello Mr Eric, I've seen your blogs and project recently.
It's very helpful to me, so I want to pay my tribute of praise for your works in first. ^^b
And I really appreciate of it.

I am working on new MIPI-DSI Panel, Innolux's product which named P101DBZ-3Z1, to attach on rpi cm3 with VC4 & KMS.

As I mentioned on title, I'm stuck at transfer timeout of "vc4 dsi"
I adapted raydium-rm68200 to innolux, p101dbz,  change match table, reset timing, init sequences but  without changing filename.
I traced into source code and I found that IRQ(vc4_dsi_irq_defer_to_thread_handler & vc4_dsi_irq_handler on drivers/gpu/drm/vc4/vc4_dsi.c) is not called.

I've working on it for weeks but I can not find the reason, could you please give me a tip?
Thanks for your efforts again.

Here are boot logs... (I add some debug messages on kernel to debug )
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.61-v7+ (pi@ubuntu) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #30 SMP Thu May 3 14:26:32 KST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Compute Module 3 Rev 1.0
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 229376
[    0.000000] free_area_init_node: node 0, pgdat 80c6f380, node_mem_map b779a000
[    0.000000]   Normal zone: 2016 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 229376 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b7753000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: cma=256M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:E4:29:E5 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=62b79e0a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 633132K/917504K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 22228K reserved, 262144K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xb8800000 - 0xff800000   (1136 MB)
    lowmem  : 0x80000000 - 0xb8000000   ( 896 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c7981c   ( 487 kB)
       .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000296] Console: colour dummy device 80x30
[    0.001204] console [tty1] enabled
[    0.001248] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001315] pid_max: default: 32768 minimum: 301
[    0.001625] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001668] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002703] Disabling memory control group subsystem
[    0.002803] CPU: Testing write buffer coherency: ok
[    0.002865] ftrace: allocating 22376 entries in 66 pages
[    0.050175] CPU0: update cpu_capacity 1024
[    0.050223] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.050284] Setting up static identity map for 0x100000 - 0x100034
[    0.052195] CPU1: update cpu_capacity 1024
[    0.052201] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.052849] CPU2: update cpu_capacity 1024
[    0.052855] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 639 lines cut ...]
[  627.694630] Workqueue: events console_callback
[  627.694644] [<8010f744>] (unwind_backtrace) from [<8010bd1c>] (show_stack+0x20/0x24)
[  627.694656] [<8010bd1c>] (show_stack) from [<80449ca0>] (dump_stack+0xc8/0x114)
[  627.694668] [<80449ca0>] (dump_stack) from [<8011cd38>] (__warn+0xe8/0x114)
[  627.694679] [<8011cd38>] (__warn) from [<8011ce34>] (warn_slowpath_null+0x30/0x38)
[  627.694691] [<8011ce34>] (warn_slowpath_null) from [<80499f64>] (clk_core_unprepare+0x16c/0x18c)
[  627.694704] [<80499f64>] (clk_core_unprepare) from [<8049b064>] (clk_unprepare+0x34/0x3c)
[  627.694776] [<8049b064>] (clk_unprepare) from [<7f25d720>] (vc4_dsi_encoder_disable+0xb8/0xc8 [vc4])
[  627.694951] [<7f25d720>] (vc4_dsi_encoder_disable [vc4]) from [<7f226ddc>] (drm_atomic_helper_commit_modeset_disables+0x420/0x4dc [drm_kms_helper])
[  627.695127] [<7f226ddc>] (drm_atomic_helper_commit_modeset_disables [drm_kms_helper]) from [<7f261698>] (vc4_atomic_complete_commit+0x30/0x8c [vc4])
[  627.695256] [<7f261698>] (vc4_atomic_complete_commit [vc4]) from [<7f2618bc>] (vc4_atomic_commit+0x1c8/0x26c [vc4])
[  627.695566] [<7f2618bc>] (vc4_atomic_commit [vc4]) from [<7f0ef660>] (drm_atomic_commit+0x5c/0x68 [drm])
[  627.695924] [<7f0ef660>] (drm_atomic_commit [drm]) from [<7f228248>] (drm_atomic_helper_connector_dpms+0xf4/0x180 [drm_kms_helper])
[  627.696144] [<7f228248>] (drm_atomic_helper_connector_dpms [drm_kms_helper]) from [<7f22bac0>] (drm_fb_helper_dpms+0xac/0x100 [drm_kms_helper])
[  627.696367] [<7f22bac0>] (drm_fb_helper_dpms [drm_kms_helper]) from [<7f22bb64>] (drm_fb_helper_blank+0x50/0xa8 [drm_kms_helper])
[  627.696486] [<7f22bb64>] (drm_fb_helper_blank [drm_kms_helper]) from [<8048bcd8>] (fb_blank+0x64/0xb8)
[  627.696498] [<8048bcd8>] (fb_blank) from [<80485eb8>] (fbcon_blank+0x1f0/0x290)
[  627.696513] [<80485eb8>] (fbcon_blank) from [<804c51f8>] (do_blank_screen+0x158/0x1f8)
[  627.696527] [<804c51f8>] (do_blank_screen) from [<804c7ea8>] (console_callback+0x84/0x164)
[  627.696541] [<804c7ea8>] (console_callback) from [<801358d8>] (process_one_work+0x144/0x438)
[  627.696556] [<801358d8>] (process_one_work) from [<80135c0c>] (worker_thread+0x40/0x574)
[  627.696570] [<80135c0c>] (worker_thread) from [<8013b8a4>] (kthread+0x108/0x124)
[  627.696583] [<8013b8a4>] (kthread) from [<80107ec8>] (ret_from_fork+0x14/0x2c)
[  627.696587] ---[ end trace 7cd1e6c5b278d8ff ]---
[  627.696599] disable_outputs: crt DISABLE b6616800

anholt, 2018-05-07:
Do you have `ignore_lcd=1` in `/boot/config.txt`?  If not, the firmware might steal interrupts before linux gets them.

jwaani2, 2018-05-08:
Thank you for your advice.
Sadly I already using ignore_lcd=1 on config.txt, but IRQ does not came up.
In fact I am not familiar device tree.
So I'm suspecting it as a device tree problem.

I add this to bcm283x.dtsi
                vc4_dsi: vc4_dsi {
                        compatible = "innolux,p101dbz-3z1", "innolux,p101dbz";
                        #address-cells = <1>;
                        #size-cells = <0>;
                        panel@0 {
                                compatible = "vc4-dsi", "vc4_dsi", "vc4 dsi";
                                reg = <0>;
                                port {
                                        panel_dsi_port: endpoint {
                                                remote-endpoint = <&dsi_out_port>;
                                        };
                                };
                        };
                };

This to dsi1 section
                        port {
                                dsi_out_port: endpoint {
                                        remote-endpoint = <&panel_dsi_port>;
                                };
                        };

And change vc4-kms-v3d-overlay.dts like this...
        fragment@15 {
                target = <&dsi1>;
                __overlay__  {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "okay";

                        pitouchscreen: panel@0 {
                                compatible = "innolux,p101dbz-3z1", "innolux,p101dbz", "raspberrypi, touchscreen";
                                reg = <0>;
                                reset-gpios = <&gpio 26 GPIO_ACTIVE_LOW>;
                                vc4_dsi = <&vc4_dsi>;

                        };
                };
        };

Please take a look these. I would be really happy if you give me some more advice.
Thanks.

EricBHLin, 2018-05-15:
Hi jwaani2,

I am doing the same thing like you, and got identical error logs. 
It seems panel initial command can't be sent.
Try to call drm_panel_prepare()  after the clock is ready in the end of vc4_dsi_encoder_enable()
it works for me, and you can get vc4_dsi_irq_handler() triggered. 

somehow, if the number of parameters are over 16 ,
you might got timeout logs like following: 
[    5.978167] dsi_dma_workaround_write:ded09000,offset=34,val=3fc0,port1
[    5.978174] packet.payload_length=33,cmd_fifo_len=1,pix_fifo_len=8
[    5.978190] dsi_dma_workaround_write:ded09000,offset=1c,val=d3,port1
[    5.978203] dsi_dma_workaround_write:ded09000,offset=20,val=40000f00,port1
[    5.978218] dsi_dma_workaround_write:ded09000,offset=20,val=3200081a,port1
[    5.978231] dsi_dma_workaround_write:ded09000,offset=20,val=8000810,port1
[    5.978244] dsi_dma_workaround_write:ded09000,offset=20,val=50f1554,port1
[    5.978257] dsi_dma_workaround_write:ded09000,offset=20,val=10120204,port1
[    5.978270] dsi_dma_workaround_write:ded09000,offset=20,val=34330705,port1
[    5.978283] dsi_dma_workaround_write:ded09000,offset=20,val=7370c0c,port1
[    5.978297] dsi_dma_workaround_write:ded09000,offset=20,val=8111707,port1
[    5.978311] dsi_dma_workaround_write:ded09000,offset=30,val=200002,port1
[    5.978324] dsi_dma_workaround_write:ded09000,offset=34,val=3fc2,port1
[    5.978337] dsi_dma_workaround_write:ded09000,offset=8,val=1002179,port1
[    5.978351] dsi_dma_workaround_write:ded09000,offset=4,val=60d,port1
[    6.792285] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    6.793255] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    7.037574] transfer interrupt wait timeout
[    7.037585] instat: 0x00000000
[    7.037620] dsi_dma_workaround_write:ded09000,offset=34,val=3fc0,port1
[    7.037714] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110

still debugging on this...
Mr. Anholt, did you have any idea on that ?

Thank you very much.

jwaani2, 2018-05-24:
Hi EricBHLin,

Thank you very very much for your help.
You were right! Now I got a IRQ triggered, after I follow your advice.

But still! there are timeouts at sending initial command.
And those timeout error's counts are changing from 1 to dozens when I reboot my board.
So I am checking the setting in vc4_dsi_encoder_enable().

Does anyone have any data for BCM2835 DSI controller's register like DSI1_HS_CLT0 ... DSI1_PHYC?
I could not find datasheet for them.
The only one that I have found is this link.
https://elinux.org/BCM2835_registers

Please somebody help me~ T.T

[    6.241574] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[    6.241584] vc4_dsi 3f700000.dsi: instat: 0x00000000
[    6.241697] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 0
[    6.241741] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[    8.321553] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[    8.321561] vc4_dsi 3f700000.dsi: instat: 0x00000000
[    8.321652] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 1
[    8.321696] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[   10.401554] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[   10.401561] vc4_dsi 3f700000.dsi: instat: 0x00000000
[   10.401655] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 2
[   10.401693] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[   12.481549] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[   12.481557] vc4_dsi 3f700000.dsi: instat: 0x00000000
[   12.481646] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 3
[   12.481684] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[   14.561546] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[   14.561553] vc4_dsi 3f700000.dsi: instat: 0x00000000
[   14.561639] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 4
[   14.561677] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[   16.641551] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[   16.641559] vc4_dsi 3f700000.dsi: instat: 0x00000000
[   16.641644] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 5
[   16.641682] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110
[   18.721552] vc4_dsi 3f700000.dsi: transfer interrupt wait timeout
[   18.721559] vc4_dsi 3f700000.dsi: instat: 0x00000000
[   18.721644] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110 6
[   18.721681] [drm:rm68200_dcs_write_buf] *ERROR* MIPI DSI DCS write buffer failed: -110

EricBHLin, 2018-05-25:
Hi jwaani2,

I can't send my initial sequence completely,too.
In my case, if the parameters in your initial commands are over 16, it will easily got these timeout errors.
But if I am lucky enough, sometime the mipi transferring can be done...(still wondering if there were hw issues)

Have you tried to read something back from the panel ? 
I think it's the easiest way to confirm DSI connection .
Despite your initial commands are sent without timeout errors, it just means these data were correctly sent to the FIFO. 

About the registers of DSI, it seems not exposed to the public
Thank you.

jwaani2, 2018-07-23:
Hi EricBHlin,

It's been a long time.

In my LCD panel, it works but there are time out errors and seems not normal.
Some times the screen looks good, but others it's broken for so many kind of cases(stripes, blanking, swapped X or Y, inverted... and so on)

For your information, I think it's better to fix arm_freq on config.txt to using DSI settled.
After I fixed the arm_freq to 600, that many error case are changed to predictable, so I could fix the problem to changing some of initial codes.

Then I finally got my LCD panel works good.
Thank you so much~ :)

jamwu, 2019-01-17:
> Hi EricBHlin,
>
> It's been a long time.
>
> In my LCD panel, it works but there are time out errors and seems not normal.
> Some times the screen looks good, but others it's broken for so many kind of [...]
@jwaani2 Hi,I'm very excited and intresting about what you have done.I want to driver a 1 lane mipi-dsi display with pi.But i am really new at DT and linux driver.Can you give me some advice?Thanks!

cappu88, 2019-02-24:
> Thank you for your advice.
> Sadly I already using ignore_lcd=1 on config.txt, but IRQ does not came up.
> In fact I am not familiar device tree.
> So I'm suspecting it as a device tree problem.
>
> I add [...]

Hi [jwaani2](/jwaani2),

i am currently trying to enable this DSI interface in the CM3 as well, however, i am not sure how to change the 3 overlay file, bcm283x.dtsi, vc4-kms-v3d-overlay.dts, and the new innolux driver that you added (maybe call raydium_rm68200.dts). Can you share this 3 overlay files to help my understanding?

bbrainka, 2019-04-05:
Hi,

Does anyone succeeded with this?