anholt/linux #135: vc4-drm soc:gpu: failed to allocate buffer with size 16777216

Joel-Mckay, 2018-07-05:
I suspect this is not a manifestation of bug  #126, but have been unable to identify exactly what causes the vc4-drm to fault and cause the HDMI screen buffer issue.  

Any insights would be helpful, as it is great for WebGL when in works....
=)

============================================
Symptoms:
Last two screen buffer states oscillate on mouse move on X.  The programs and system continue to run even though the display has locked up.  

============================================
Excluding other sources of possible issues:
1.) I disabled the wifi and bluetooth, as the the corruption initially seemed to occur if the wifi card was not associated with a base-station. 
This was not related to the issue, but did seem to increase stability to 14 hours. The kernel sometimes warns of various tainted programs, but the cma settings do not seem to change the bug manifestation delay.

2.) I left the display on the LightDM interface as I thought it may be related to the compton GLX mode in MATE/Marco.  I also tried the packagers version of the regular compton.conf.
This did not fix to the issue.

3.) Switched back from git 4.14.34-rt27-v7+ to the repo kernel 4.14.37-v7+.img
This did not fix to the issue.

4.) Switched back from repo firmware images to   rpi-update Hexxeh rpi-4.14.y images.
This did not fix to the issue.

============================================
Affected platforms (the odd part is these should all behave the same):
//hangs the display within 8 hours
Hardware	: BCM2835
Revision	: a020d3
a020d3 	Q1 2018 	3 Model B+ 	1.3 	1 GB 	(Mfg by Sony) 

//hangs the display within 24 hours
Hardware	: BCM2835
Revision	: a22082
a22082 	Q1 2016 	3 Model B 	1.2 	1 GB 	(Mfg by Embest)

//Seems stable after 6 days with the identical sdcard/settings (very odd)
Revision	: a21041
a21041 	Q1 2015 	2 Model B 	1.1 	1 GB 	(Mfg by Embest)

============================================
/boot/cmdline.txt
dwc_otg.fiq_fsm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.nak_holdoff=0 dwc_otg.lpm_enable=0  logo.nologo console=tty3 kgdboc=tty3 loglevel=3 vt.global_cursor_default=0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait usbcore.old_scheme_first=N coherent_pool=6M smsc95xx.turbo_mode=N quiet plymouth.ignore-serial-consoles video=1024x768

============================================
/boot/config.txt
#tried repo Raspbian Stretch kernel
kernel=4.14.37-v7+.img
#and the RT kernel branch
#kernel=4.14.34-rt27-v7+.img
hdmi_safe=1
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_force_edid_audio=1
hdmi_group=2
hdmi_mode=16
config_hdmi_boost=4
framebuffer_width=1024
framebuffer_height=768
gpu_mem=128
cma_lwm=64
cma_hwm=96
avoid_safe_mode=1
disable_splash=1
temp_limit=76
max_usb_current=1
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=i2c_arm_baudrate=100000
dtparam=spi=on
dtparam=random=on
[... 323 lines cut ...]
Jul  5 01:54:30 neptune kernel: [28485.885612] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jul  5 01:54:30 neptune kernel: [28485.885634] brcmfmac: power management disabled
Jul  5 01:59:46 neptune kernel: [28801.845512] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jul  5 01:59:46 neptune kernel: [28801.845533] brcmfmac: power management disabled
Jul  5 02:05:02 neptune kernel: [29117.701794] ------------[ cut here ]------------
Jul  5 02:05:02 neptune kernel: [29117.701927] WARNING: CPU: 1 PID: 2550 at drivers/gpu/drm/vc4/vc4_crtc.c:668 vc4_crtc_atomic_flush+0x124/0x128 [vc4]
Jul  5 02:05:02 neptune kernel: [29117.701957] Modules linked in: cmac rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic binfmt_misc joydev evdev hid_microsoft spidev brcmfmac brcmutil cfg80211 vc4 rfkill drm_kms_helper cec drm snd_soc_core snd_compress snd_bcm2835(C) snd_pcm_dmaengine syscopyarea snd_pcm sysfillrect snd_timer sysimgblt fb_sys_fops snd spi_bcm2835 i2c_bcm2835 uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Jul  5 02:05:02 neptune kernel: [29117.702084] CPU: 1 PID: 2550 Comm: kworker/u8:2 Tainted: G        WC      4.14.34-rt27-v7+ #1
Jul  5 02:05:02 neptune kernel: [29117.702087] Hardware name: BCM2835
Jul  5 02:05:02 neptune kernel: [29117.702154] Workqueue: events_unbound commit_work [vc4]
Jul  5 02:05:02 neptune kernel: [29117.702187] [<8010fe10>] (unwind_backtrace) from [<8010bf6c>] (show_stack+0x20/0x24)
Jul  5 02:05:02 neptune kernel: [29117.702204] [<8010bf6c>] (show_stack) from [<807b42ac>] (dump_stack+0xbc/0x100)
Jul  5 02:05:02 neptune kernel: [29117.702220] [<807b42ac>] (dump_stack) from [<8011dc1c>] (__warn+0xf8/0x110)
Jul  5 02:05:02 neptune kernel: [29117.702232] [<8011dc1c>] (__warn) from [<8011dd04>] (warn_slowpath_null+0x30/0x38)
Jul  5 02:05:02 neptune kernel: [29117.702301] [<8011dd04>] (warn_slowpath_null) from [<7f338ef0>] (vc4_crtc_atomic_flush+0x124/0x128 [vc4])
Jul  5 02:05:02 neptune kernel: [29117.702498] [<7f338ef0>] (vc4_crtc_atomic_flush [vc4]) from [<7f2dbab0>] (drm_atomic_helper_commit_planes+0x1c4/0x2b0 [drm_kms_helper])
Jul  5 02:05:02 neptune kernel: [29117.702691] [<7f2dbab0>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<7f340304>] (vc4_atomic_complete_commit+0x54/0xd8 [vc4])
Jul  5 02:05:02 neptune kernel: [29117.702815] [<7f340304>] (vc4_atomic_complete_commit [vc4]) from [<7f3404d4>] (commit_work+0x1c/0x20 [vc4])
Jul  5 02:05:02 neptune kernel: [29117.702889] [<7f3404d4>] (commit_work [vc4]) from [<80138408>] (process_one_work+0x1fc/0x588)
Jul  5 02:05:02 neptune kernel: [29117.702905] [<80138408>] (process_one_work) from [<8013942c>] (worker_thread+0x60/0x5e4)
Jul  5 02:05:02 neptune kernel: [29117.702923] [<8013942c>] (worker_thread) from [<8013ebc8>] (kthread+0x144/0x174)
Jul  5 02:05:02 neptune kernel: [29117.702941] [<8013ebc8>] (kthread) from [<80107d74>] (ret_from_fork+0x14/0x20)
Jul  5 02:05:02 neptune kernel: [29117.702997] ---[ end trace 0000000000000004 ]---
Jul  5 02:05:02 neptune kernel: [29117.837425] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jul  5 02:05:02 neptune kernel: [29117.837443] brcmfmac: power management disabled

Joel-Mckay, 2018-07-05:
@stschake 
Thanks, I was wondering why someone hasn't jettisoned this feature into space yet ;-) 

//I have started to re-build the head from 3 days ago:
git clone https://github.com/raspberrypi/linux.git  -b "rpi-4.14.y-rt" --depth 1
cd linux

//which branch/patch did you use to fix the cma thrashing bug on?
//todo: git merge https://github.com/anholt/linux.git  -b "rpi-4.14.y"

//I also checked the kernel settings made by the default setup template:
sudo modprobe configs
cp /proc/config.gz ./
zcat config.gz > .config

nano  .config
CONFIG_DMA_CMA=y
///changed sctipt default: CONFIG_CMA_SIZE_MBYTES=5
//as per: https://github.com/shirishpargaonkar/sspcifso/commit/ebf089248dab2ef569e5e26a607f0977a71182b7
CONFIG_CMA_SIZE_MBYTES=32

//currently running
make -j4 zImage modules dtbs

6by9, 2018-07-05:
Why the change to the config?

The size of the CMA heap is specified on the command line, or via device tree.
```cma=256M``` is present in your cmdline.txt, therefore you should be running with 256MB.
The vc4-kms-v3d overlay also defaults to 256M. https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/overlays/vc4-kms-v3d-overlay.dts#L16

Confirm by looking in the kernel log straight after boot. You should get a line similar to
```
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
```

lategoodbye, 2018-07-05:
@6by9 FWIW, the reason for this defconfig change was the inflexibility of kernelci.org. They don't want platform specific kernel parameter.

Joel-Mckay, 2018-07-05:
@6by9 
Indeed, it was my futile attempt to setup various contiguous blocks of memory to figure out why the fault was being generated (did not change the event times).  stschake's patch does sound interesting given it may solve 2 different failure modes with long running systems.   The part I can't figure out is why a pi2 doesn't fault the same way running the identical software/kernel....

Thus, I am preparing a new kernel to see if: 
1. The issue(s) were already fixed in the head merge to the RT kernel 3 days ago (unlikely, as you've stated, but our conversation will be about current builds)
2. I am curious if the cited patch should be in the "make bcm2709_defconfig" scripts
3. When stschake's specific cma patch is listed it can be merged and tested (on 2B / 3B / 3B+)

I will update the log as tests are completed, and I am certainly open to suggestions.

stschake, 2018-07-05:
You can try this patch, written from memory:

[0001-drm-vc4-Keep-the-binner-BO-through-suspend.patch.txt](https://github.com/anholt/linux/files/2167982/0001-drm-vc4-Keep-the-binner-BO-through-suspend.patch.txt)

Can you confirm you are running with 256 MiB CMA? Check the kernel log, the default kernel configuration setting might or might not apply.

Joel-Mckay, 2018-07-05:
I am currently testing: 
kernel=4.14.52-rt31-rc1-v7+.img
dtoverlay=vc4-kms-v3d,cma-128
avoid_warnings=2
cma_offline_start=128
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_force_edid_audio=1
hdmi_group=2
hdmi_mode=16
config_hdmi_boost=4
overscan_left=10
overscan_right=10
overscan_top=10
overscan_bottom=10
disable_overscan=0
console-width=1024
console-height=700
gpu_mem=256
gpu_mem_256=256
gpu_mem_512=256
gpu_mem_1024=256
#tried to reduce thrashing here
cma_lwm=128
cma_hwm=128
avoid_safe_mode=1
disable_splash=1
temp_limit=76
max_usb_current=1
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=i2c_arm_baudrate=100000
dtparam=spi=on
dtparam=random=on
dtparam=watchdog=on
enable_uart=1

6by9, 2018-07-05:
That's a fair number of random entries in config.txt
```
cma_lwm=64
cma_hwm=96
cma_offline_start=128
```
should all do nothing - the driver was removed a while back as it was never reliable. https://github.com/raspberrypi/linux/pull/2233 and https://github.com/raspberrypi/firmware/commit/cab026ba87e1a25b5673ecb658a62e30fb6537c9

```gpu_mem_256=256``` is going to cause issues on a 256MB Pi as you've just assigned all the memory to the VPU so the ARM gets none. 256MiB is excessive anyway for almost all use cases (the camera is normally the heaviest weight use case and is more than happy on 128).

All of the overscan and hdmi settings are irrelevant as you're using the vc4-kms-v3d driver.
I can't recall if you need ```avoid_warnings=2``` to disable the over-temp and under voltage warnings. I vaguely recall that the firmware disables those automatically if the vc4 driver is loaded.

The rest looks vaguely sane, although give vc4-kms-v3d the normal 256MiB of CMA via ```dtoverlay=vc4-kms-v3d``` unless you have a very good reason to do otherwise.

Joel-Mckay, 2018-07-05:
@stschake 
The error no longer seems to hang X with your patch to drivers/gpu/drm/vc4/vc4_v3d.c
Currently testing on pi 3b+, and will let you know how things manifest in the next day.
Thanks again,
=)

Jul  5 20:13:52 neptune kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jul  5 20:13:52 neptune kernel: [    0.000000] Linux version 4.14.52-rt31-rc1-v7+ (pi@neptune) (gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)) #1 SMP PREEMPT RT Thu Jul 5 15:00:38 UTC 2018
Jul  5 20:13:52 neptune kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jul  5 20:13:52 neptune kernel: [    0.000000] CPU: div instructions available: patching division code
Jul  5 20:13:52 neptune kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Jul  5 20:13:52 neptune kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
Jul  5 20:13:52 neptune kernel: [    0.000000] Memory policy: Data cache writealloc
Jul  5 20:13:52 neptune kernel: [    0.000000] cma: Reserved 128 MiB at 0x26c00000
Jul  5 20:13:52 neptune kernel: [    0.000000] On node 0 totalpages: 196608
Jul  5 20:13:52 neptune kernel: [    0.000000] free_area_init_node: node 0, pgdat 80c8a980, node_mem_map af8fa000
Jul  5 20:13:52 neptune kernel: [    0.000000]   Normal zone: 1728 pages used for memmap
Jul  5 20:13:52 neptune kernel: [    0.000000]   Normal zone: 0 pages reserved
Jul  5 20:13:52 neptune kernel: [    0.000000]   Normal zone: 196608 pages, LIFO batch:31
Jul  5 20:13:52 neptune kernel: [    0.000000] percpu: Embedded 18 pages/cpu @af89f000 s41440 r8192 d24096 u73728
Jul  5 20:13:52 neptune kernel: [    0.000000] pcpu-alloc: s41440 r8192 d24096 u73728 alloc=18*4096
Jul  5 20:13:52 neptune kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Jul  5 20:13:52 neptune kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
Jul  5 20:13:52 neptune kernel: [    0.000000] Kernel command line: 8250.nr_uarts=1 cma=128M bcm2708_fb.fbwidth=1004 bcm2708_fb.fbheight=748 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0  logo.nologo console=tty3 kgdboc=tty3 loglevel=3 vt.global_cursor_default=0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait usbcore.old_scheme_first=N coherent_pool=8M smsc95xx.turbo_mode=N quiet plymouth.ignore-serial-consoles video=1024x768
Jul  5 20:13:52 neptune kernel: [    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
Jul  5 20:13:52 neptune kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
Jul  5 20:13:52 neptune kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Jul  5 20:13:52 neptune kernel: [    0.000000] Memory: 633836K/786432K available (7168K kernel code, 581K rwdata, 2108K rodata, 1024K init, 750K bss, 21524K reserved, 131072K cma-reserved)
Jul  5 20:13:52 neptune kernel: [    0.000000] Virtual kernel memory layout:
Jul  5 20:13:52 neptune kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000]     vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
Jul  5 20:13:52 neptune kernel: [    0.000000]     lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
Jul  5 20:13:52 neptune kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
Jul  5 20:13:52 neptune kernel: [    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000]       .data : 0x80c00000 - 0x80c915c0   ( 582 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000]        .bss : 0x80c992d0 - 0x80d54e9c   ( 751 kB)
Jul  5 20:13:52 neptune kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jul  5 20:13:52 neptune kernel: [    0.000000] ftrace: allocating 25367 entries in 75 pages
Jul  5 20:13:52 neptune kernel: [    0.000000] Preemptible hierarchical RCU implementation.
Jul  5 20:13:52 neptune kernel: [    0.000000] 	No expedited grace period (rcu_normal_after_boot).
Jul  5 20:13:52 neptune kernel: [    0.000000] 	Tasks RCU enabled.
Jul  5 20:13:52 neptune kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jul  5 20:13:52 neptune kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Jul  5 20:13:52 neptune kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Jul  5 20:13:52 neptune kernel: [    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Jul  5 20:13:52 neptune kernel: [    0.000016] Switching to timer-based delay loop, resolution 52ns
Jul  5 20:13:52 neptune kernel: [    0.000260] Console: colour dummy device 80x30
Jul  5 20:13:52 neptune kernel: [    0.000282] console [tty3] enabled
Jul  5 20:13:52 neptune kernel: [    0.000306] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=38400)
Jul  5 20:13:52 neptune kernel: [    0.000318] pid_max: default: 32768 minimum: 301
Jul  5 20:13:52 neptune kernel: [    0.000631] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[... cut ...]
Jul  5 20:23:44 neptune kernel: [  588.709084] [drm]                     V3D shader:    320kb BOs (74)
Jul  5 20:23:44 neptune kernel: [  588.709087] [drm]                           dumb:     48kb BOs (3)
Jul  5 20:23:44 neptune kernel: [  588.709090] [drm]                         binner:  16384kb BOs (1)
Jul  5 20:23:44 neptune kernel: [  588.712439] vc4-drm soc:gpu: failed to allocate buffer with size 1556480
Jul  5 20:25:07 neptune kernel: [  672.554419] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Jul  5 20:25:07 neptune kernel: [  672.554434] brcmfmac: power management disabled

Joel-Mckay, 2018-07-07:
@6by9 
Thanks for tips, as I have also been trying to get the mp4 codec support in "browser-plugin-vlc" package by patching and linking on the OpenGL (pi patch) enabled libVLC:
//edited: npapi-vlc/npapi/vlcplugin_base.cpp
    ppsz_argv[ppsz_argc++] = "--vout=xcb_glx"; 
//Note, this didn't seem to alter software CPU loads in midori even though VLC itself does 720p easily with the hardware codec.  

The webGL seems to flake out a bit during initial use, but Midori is so unstable normally I can't tell if its related to  the cma mmu bypass handling issue.
//The normal mode testing of vc4: /boot/config.txt
kernel=4.14.52-rt31-rc1-v7+.img
dtoverlay=vc4-kms-v3d
avoid_warnings=2
hdmi_force_hotplug=1
hdmi_drive=2
hdmi_force_edid_audio=1
hdmi_group=2
hdmi_mode=16
config_hdmi_boost=4
disable_overscan=0
console-width=1024
console-height=700
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
gpu_mem_1024=128
avoid_safe_mode=1
disable_splash=1
temp_limit=76
max_usb_current=1
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=i2c_arm_baudrate=100000
dtparam=spi=on
dtparam=random=on
dtparam=watchdog=on
enable_uart=1

@stschake
Thank you for posting the patch to at least get this opengl browser working.
You were correct in that the patch does seems to reduce error rates the longer it runs.

I still have no idea why the disk image running on Pi2 (Revision: a21041)
That uses the same vc4  blob as the pi3 running stretch: 0.0.0 20140616
Doesn't exhibit the same memory issue yet even with the vanilla kernel on an LG LCD: uptime: 10:22:13 up 10 days, 23:10,  2 users,  load average: 0.07, 0.10, 0.14