anholt/linux #137: 1600x1200 not working with VC4 Full KMS

scpcom, 2018-07-30:
My Raspberry Pi 3B is connected to a hp 2035 LCD monitor which has a native resolution of 1600x1200 60 Hz (pixel clock 162000).
In legacy mode it works, in VC4 Full KMS mode the resolution goes down to 1280x960.

This was caused by this commit:
https://github.com/raspberrypi/linux/commit/48466df61a9c226b1b51e361aedf977dea6baf5c
If I compile a kernel with (1000 * 101 / 100) instead of (1000 * 108 / 100) it works perfectly.
es2gears and glxgears runs with 60 fps.
Where does the definition "HSM clock must be 108% of the pixel clock" come from?

anholt, 2018-07-30:
It comes from the hardware specification.  I would bet that something like HDMI audio or CEC would be the first things to break when running the HDMI pixel clock too high for the state machine.

anholt, 2018-07-30:
Possibly infoframe packets, too.  We'd need a hardware engineer to answer the question.

scpcom, 2018-07-30:
Hmm, that would make sense.
Maybe we can get around it by using a reduced blanking (rb) mode. I will try around to see if there is another solution that is good enough to get in the kernel tree.
I do not want to compile a custom kernel just for a simple standard resolution :-)

scpcom, 2018-08-02:
I remove the old workaround and added this quirk:
http://seafile.servator.de/sbc/raspberrypi/build/linux-4.14.y-drm-vc4-pixel_clock_quirk.patch
And it works. That maybe a possible solution.
For 1600x1200@60 a CVT mode with pixel clock 130.32 MHz will be created and used.
The quirk also applies to all other resolutions with a clock higher than 151.55 MHz.