anholt/linux #140: vc4: NULL pointer dereference in drm_client_dev_hotplug

lategoodbye, 2018-09-28:
@orpaltech [reported](https://lists.freedesktop.org/archives/dri-devel/2018-September/188812.html) this NULL pointer dereference at dri-devel. Now he was able to provide a Kernel config and i'm able to reproduce it with a Raspberry Pi 3 (arm64) and Linux 4.19-rc5. At seems like a invalid config for VC4, but nevertheless the driver shouldn't crash.

```
[    9.124598] vc4_hdmi 3f902000.hdmi: ASoC: Failed to create component debugfs directory
[    9.147667] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi mapping ok
[    9.155184] vc4_hdmi 3f902000.hdmi: ASoC: no DMI vendor name!
[    9.166544] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[    9.173840] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[    9.181029] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[    9.188519] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    9.195690] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.203523] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.215032] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.274785] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[    9.290246] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[    9.297464] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    9.304600] [drm] Driver supports precise vblank timestamp query.
[    9.323839] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    9.382856] vc4-drm soc:gpu: [drm:drm_fb_helper_fbdev_setup [drm_kms_helper]] *ERROR* Failed to set fbdev configuration
[    9.513495] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    9.522507] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[    9.532939] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    9.545064] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   10.404937] Unable to handle kernel paging request at virtual address 00330a656369768a
[   10.413081] Mem abort info:
[   10.415939]   ESR = 0x96000004
[   10.419082]   Exception class = DABT (current EL), IL = 32 bits
[   10.425357]   SET = 0, FnV = 0
[   10.428533]   EA = 0, S1PTW = 0
[   10.431729] Data abort info:
[   10.434674]   ISV = 0, ISS = 0x00000004
[   10.438589]   CM = 0, WnR = 0
[   10.441620] [00330a656369768a] address between user and kernel address ranges
[   10.449087] Internal error: Oops: 96000004 [#1] PREEMPT SMP
[   10.454762] Modules linked in: brcmfmac vc4 drm_kms_helper cfg80211 drm rfkill smsc95xx brcmutil usbnet drm_panel_orientation_quirks raspberrypi_hwmon bcm2835_dma crc32_ce pwm_bcm2835 bcm2835_rng virt_dma rng_core i2c_bcm2835 ip_tables x_tables ipv6
[   10.477296] CPU: 2 PID: 45 Comm: kworker/2:1 Not tainted 4.19.0-rc5 #3
[   10.483934] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[   10.489966] Workqueue: events output_poll_execute [drm_kms_helper]
[   10.496261] pstate: 80000005 (Nzcv daif -PAN -UAO)
[   10.501321] pc : drm_client_dev_hotplug+0x5c/0xb0 [drm]
[   10.506776] lr : drm_client_dev_hotplug+0x38/0xb0 [drm]
[   10.512087] sp : ffff000008e83cd0
[   10.515455] x29: ffff000008e83cd0 x28: 0000000000000000 
[   10.520859] x27: ffff000000ad1c60 x26: ffff800034207230 
[   10.526263] x25: 0000000000000001 x24: 0000000000000002 
[   10.531666] x23: ffff8000342070f8 x22: ffff0000009bd398 
[   10.537070] x21: ffff800034207118 x20: ffff800034207000 
[   10.542474] x19: ffff800035638800 x18: ffffffffffffffff 
[   10.547877] x17: 0000000000000000 x16: 0000000000000000 
[   10.553281] x15: 0000000000000400 x14: 0000000000000400 
[   10.558685] x13: 0000000000000002 x12: 0000000000000000 
[   10.564088] x11: 0000000000000000 x10: 0000000000000000 
[   10.569492] x9 : 0000000000000001 x8 : 00000000ffffffff 
[   10.574896] x7 : ffff8000373a52a0 x6 : ffff800033087500 
[   10.580298] x5 : ffff000008e83ba0 x4 : ffff8000373bdc20 
[   10.585701] x3 : ffff8000342070f8 x2 : ffff800035fb9a80 
[   10.591104] x1 : 30330a6563697672 x0 : ffff800035638800 
[   10.596515] Process kworker/2:1 (pid: 45, stack limit = 0x000000007e8924dc)
[   10.603590] Call trace:
[   10.606259]  drm_client_dev_hotplug+0x5c/0xb0 [drm]
[   10.611303]  drm_kms_helper_hotplug_event+0x30/0x40 [drm_kms_helper]
[   10.617849]  output_poll_execute+0xc4/0x1e0 [drm_kms_helper]
[   10.623616]  process_one_work+0x1c8/0x318
[   10.627695]  worker_thread+0x48/0x428
[   10.631420]  kthread+0xf8/0x128
[   10.634615]  ret_from_fork+0x10/0x18
[   10.638255] Code: 54000220 f9401261 aa1303e0 b4000141 (f9400c21) 
[   10.644456] ---[ end trace c75b4a4b0e141908 ]---
```

sergey-suloev, 2018-09-28:
I confirm that default arm64 config  works well (with no crash). I am trying to compare it with my config to find possible reasons.

lategoodbye, 2018-10-09:
This issue has been fixed:
https://patchwork.freedesktop.org/patch/254272/

closed 2018-10-09 by lategoodbye