anholt/linux #141: Support for screens larger than 2048x2048

jiridurdis, 2018-10-04:
Hi,

I believe that while Pi is suitable for video playback up to 1080p only, it still would be nice to support larger screens for other not-so-demanding scenarios. At the moment, maximal screen size is hardcoded to 2048x2048 in vc4_kms.c.

Pi is capable of 4K (3840x2160@25Hz) output over HDMI (using fbturbo and custom hdmi_cvt or hdmi_timings line in config.txt), which is totally OK for displaying dashboards, demos, slideshows etc.

anholt, 2018-10-16:
Care to submit a tested patch to the dri-devel mailing list?  I don't have 4k hw to test against.

jiridurdis, 2018-10-16:
Hi,

Simple change of values in vc4_kms.c to 4096x4096 is not sufficient to make it work.

It allows me to create a new custom resolution using `xrandr`, but it fails when I try to apply it.
This works with 4096x4096 as new maximal screen size:
```
xrandr --newmode "3840x2160"  220.43  3840 3888 3920 4000  2160 2163 2168 2222 +hsync +vsync
xrandr --addmode HDMI-1 "3840x2160"
```

But this fails with error:
```
xrandr --output HDMI-1 --mode "3840x2160" --verbose
```
```
screen 0: 3840x2160 1013x570 mm  96.25dpi
crtc 2:  3840x2160_1  25.00 +0+0 "HDMI-1"
xrandr: Configure crtc 2 failed
crtc 0: disable
crtc 1: disable
crtc 2: disable
screen 0: revert
crtc 0: revert
crtc 1: revert
crtc 2: revert
```

This modeline is working for me with fbturbo_dev and increased hdmi_pixel_freq_limit.

Any idea on why this is failing?

jiridurdis, 2018-10-16:
That could be it, thank you for suggestion. Will try tomorrow.

jiridurdis, 2018-10-17:
After increasing pixel clock limit, I can apply that modeline, but it crashes driver and X server with following output in dmesg:
```
[   33.581666] ------------[ cut here ]------------
[   33.581762] WARNING: CPU: 2 PID: 621 at drivers/gpu/drm/drm_atomic_helper.c:1240 drm_atomic_helper_wait_for_vblanks.part.0+0x260/0x26c [drm_kms_helper]
[   33.581766] [CRTC:67:crtc-2] vblank wait timed out
[   33.581769] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6 overlay vc4 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd i2c_bcm2835 fixed
[   33.581860] CPU: 2 PID: 621 Comm: Xorg Tainted: G         C      4.14.70-v7 #1
[   33.581863] Hardware name: BCM2835
[   33.581883] [<8010fb70>] (unwind_backtrace) from [<8010bee8>] (show_stack+0x20/0x24)
[   33.581893] [<8010bee8>] (show_stack) from [<8076dbc0>] (dump_stack+0xc8/0x114)
[   33.581902] [<8076dbc0>] (dump_stack) from [<8011d464>] (__warn+0xe8/0x114)
[   33.581910] [<8011d464>] (__warn) from [<8011d4d8>] (warn_slowpath_fmt+0x48/0x50)
[   33.581977] [<8011d4d8>] (warn_slowpath_fmt) from [<7f162eac>] (drm_atomic_helper_wait_for_vblanks.part.0+0x260/0x26c [drm_kms_helper])
[   33.582107] [<7f162eac>] (drm_atomic_helper_wait_for_vblanks.part.0 [drm_kms_helper]) from [<7f162edc>] (drm_atomic_helper_wait_for_vblanks+0x24/0x28 [drm_kms_helper])
[   33.582210] [<7f162edc>] (drm_atomic_helper_wait_for_vblanks [drm_kms_helper]) from [<7f19ff14>] (vc4_atomic_complete_commit+0x88/0xd8 [vc4])
[   33.582283] [<7f19ff14>] (vc4_atomic_complete_commit [vc4]) from [<7f1a003c>] (vc4_atomic_commit+0xd8/0x124 [vc4])
[   33.582469] [<7f1a003c>] (vc4_atomic_commit [vc4]) from [<7f0cf740>] (drm_atomic_commit+0x5c/0x68 [drm])
[   33.582677] [<7f0cf740>] (drm_atomic_commit [drm]) from [<7f1639c4>] (drm_atomic_helper_set_config+0x94/0xb4 [drm_kms_helper])
[   33.582874] [<7f1639c4>] (drm_atomic_helper_set_config [drm_kms_helper]) from [<7f0c3598>] (__drm_mode_set_config_internal+0x68/0xec [drm])
[   33.583129] [<7f0c3598>] (__drm_mode_set_config_internal [drm]) from [<7f0c3edc>] (drm_mode_setcrtc+0x3a8/0x4f4 [drm])
[   33.583383] [<7f0c3edc>] (drm_mode_setcrtc [drm]) from [<7f0be7e8>] (drm_ioctl_kernel+0x78/0xb4 [drm])
[   33.583635] [<7f0be7e8>] (drm_ioctl_kernel [drm]) from [<7f0beb2c>] (drm_ioctl+0x194/0x34c [drm])
[   33.583770] [<7f0beb2c>] (drm_ioctl [drm]) from [<80296654>] (do_vfs_ioctl+0x9c/0x7c0)
[   33.583779] [<80296654>] (do_vfs_ioctl) from [<80296dbc>] (SyS_ioctl+0x44/0x6c)
[   33.583789] [<80296dbc>] (SyS_ioctl) from [<80107dc0>] (ret_fast_syscall+0x0/0x28)
[   33.583795] ---[ end trace ad75b12d34bcd044 ]---
[   43.993392] [drm:drm_atomic_helper_commit_cleanup_done [drm_kms_helper]] *ERROR* [CRTC:67:crtc-2] flip_done timed out
[   48.770042] vc4-drm soc:gpu: failed to allocate buffer with size 14680064
```

Last line with error about buffer allocation was there after second attempt to apply that modeline, so it may not be related to this problem.

lategoodbye, 2018-10-17:
@jiridurdis Looks like you are using the Foundation kernel, are you able to build a recent mainline version like 4.19-rc8 or at least 4.18?

jiridurdis, 2018-10-17:
Still crashing on vanilla 4.18.14:
```
[  232.702997] ------------[ cut here ]------------
[  232.703084] WARNING: CPU: 3 PID: 8039 at drivers/gpu/drm/drm_atomic_helper.c:1362 drm_atomic_helper_wait_for_vblanks.part.1+0x23c/0x248 [drm_kms_helper]
[  232.703089] [CRTC:89:crtc-2] vblank wait timed out
[  232.703093] Modules linked in: cmac brcmfmac sha256_generic sha256_arm cfg80211 brcmutil crc32_arm_ce hci_uart btbcm bluetooth ecdh_generic bcm2835_thermal vc4 snd_soc_core ac97_bus snd_pcm_dmaengine snd_pcm snd_timer snd soundcore cec drm_kms_helper drm phy_generic
[  232.703184] CPU: 3 PID: 8039 Comm: Xorg Not tainted 4.18.14 #2
[  232.703188] Hardware name: BCM2835
[  232.703219] [<c0311b04>] (unwind_backtrace) from [<c030caec>] (show_stack+0x10/0x14)
[  232.703239] [<c030caec>] (show_stack) from [<c0cd622c>] (dump_stack+0x88/0xa4)
[  232.703257] [<c0cd622c>] (dump_stack) from [<c0344d20>] (__warn+0xe4/0x110)
[  232.703271] [<c0344d20>] (__warn) from [<c0344d84>] (warn_slowpath_fmt+0x38/0x48)
[  232.703330] [<c0344d84>] (warn_slowpath_fmt) from [<bf0802d4>] (drm_atomic_helper_wait_for_vblanks.part.1+0x23c/0x248 [drm_kms_helper])
[  232.703434] [<bf0802d4>] (drm_atomic_helper_wait_for_vblanks.part.1 [drm_kms_helper]) from [<bf14a4e4>] (vc4_atomic_complete_commit+0xac/0x350 [vc4])
[  232.703531] [<bf14a4e4>] (vc4_atomic_complete_commit [vc4]) from [<bf14a8b8>] (vc4_atomic_commit+0x130/0x160 [vc4])
[  232.703627] [<bf14a8b8>] (vc4_atomic_commit [vc4]) from [<bf080c78>] (drm_atomic_helper_set_config+0x6c/0x98 [drm_kms_helper])
[  232.703805] [<bf080c78>] (drm_atomic_helper_set_config [drm_kms_helper]) from [<bf011bf4>] (__drm_mode_set_config_internal+0x58/0xec [drm])
[  232.704033] [<bf011bf4>] (__drm_mode_set_config_internal [drm]) from [<bf012528>] (drm_mode_setcrtc+0x428/0x53c [drm])
[  232.704254] [<bf012528>] (drm_mode_setcrtc [drm]) from [<bf00d0f8>] (drm_ioctl_kernel+0x84/0xc0 [drm])
[  232.704473] [<bf00d0f8>] (drm_ioctl_kernel [drm]) from [<bf00d4ac>] (drm_ioctl+0x1d4/0x3c0 [drm])
[  232.704594] [<bf00d4ac>] (drm_ioctl [drm]) from [<c047780c>] (do_vfs_ioctl+0x88/0x8fc)
[  232.704608] [<c047780c>] (do_vfs_ioctl) from [<c04780b4>] (ksys_ioctl+0x34/0x5c)
[  232.704621] [<c04780b4>] (ksys_ioctl) from [<c0301000>] (ret_fast_syscall+0x0/0x54)
[  232.704628] Exception stack(0xd3969fa8 to 0xd3969ff0)
[  232.704639] 9fa0:                   b6fc0ce8 bef415a0 0000000c c06864a2 bef415a0 00000001
[  232.704651] 9fc0: b6fc0ce8 bef415a0 c06864a2 00000036 00be3e30 00000091 00000059 00000000
[  232.704659] 9fe0: b6dc708c bef41584 b6dae950 b6aab8ac
[  232.704668] ---[ end trace 675cb1b5b812c856 ]---
[  232.762986] ------------[ cut here ]------------
[  232.763068] WARNING: CPU: 3 PID: 8039 at drivers/gpu/drm/drm_atomic_helper.c:1362 drm_atomic_helper_wait_for_vblanks.part.1+0x23c/0x248 [drm_kms_helper]
[  232.763073] [CRTC:89:crtc-2] vblank wait timed out
[  232.763077] Modules linked in: cmac brcmfmac sha256_generic sha256_arm cfg80211 brcmutil crc32_arm_ce hci_uart btbcm bluetooth ecdh_generic bcm2835_thermal vc4 snd_soc_core ac97_bus snd_pcm_dmaengine snd_pcm snd_timer snd soundcore cec drm_kms_helper drm phy_generic
[  232.763169] CPU: 3 PID: 8039 Comm: Xorg Tainted: G        W         4.18.14 #2
[  232.763173] Hardware name: BCM2835
[  232.763204] [<c0311b04>] (unwind_backtrace) from [<c030caec>] (show_stack+0x10/0x14)
[  232.763222] [<c030caec>] (show_stack) from [<c0cd622c>] (dump_stack+0x88/0xa4)
[  232.763240] [<c0cd622c>] (dump_stack) from [<c0344d20>] (__warn+0xe4/0x110)
[  232.763254] [<c0344d20>] (__warn) from [<c0344d84>] (warn_slowpath_fmt+0x38/0x48)
[  232.763314] [<c0344d84>] (warn_slowpath_fmt) from [<bf0802d4>] (drm_atomic_helper_wait_for_vblanks.part.1+0x23c/0x248 [drm_kms_helper])
[  232.763417] [<bf0802d4>] (drm_atomic_helper_wait_for_vblanks.part.1 [drm_kms_helper]) from [<bf14a4e4>] (vc4_atomic_complete_commit+0xac/0x350 [vc4])
[  232.763506] [<bf14a4e4>] (vc4_atomic_complete_commit [vc4]) from [<bf14a8b8>] (vc4_atomic_commit+0x130/0x160 [vc4])
[  232.763674] [<bf14a8b8>] (vc4_atomic_commit [vc4]) from [<bf01e168>] (drm_atomic_connector_commit_dpms+0xe8/0xf8 [drm])
[  232.763905] [<bf01e168>] (drm_atomic_connector_commit_dpms [drm]) from [<bf023678>] (drm_mode_obj_set_property_ioctl+0x18c/0x284 [drm])
[  232.764133] [<bf023678>] (drm_mode_obj_set_property_ioctl [drm]) from [<bf022174>] (drm_mode_connector_property_set_ioctl+0x34/0x40 [drm])
[  232.764357] [<bf022174>] (drm_mode_connector_property_set_ioctl [drm]) from [<bf00d0f8>] (drm_ioctl_kernel+0x84/0xc0 [drm])
[  232.764576] [<bf00d0f8>] (drm_ioctl_kernel [drm]) from [<bf00d4ac>] (drm_ioctl+0x1d4/0x3c0 [drm])
[  232.764695] [<bf00d4ac>] (drm_ioctl [drm]) from [<c047780c>] (do_vfs_ioctl+0x88/0x8fc)
[  232.764708] [<c047780c>] (do_vfs_ioctl) from [<c04780b4>] (ksys_ioctl+0x34/0x5c)
[  232.764721] [<c04780b4>] (ksys_ioctl) from [<c0301000>] (ret_fast_syscall+0x0/0x54)
[  232.764728] Exception stack(0xd3969fa8 to 0xd3969ff0)
[  232.764738] 9fa0:                   00000000 bef415f0 0000000c c01064ab bef415f0 7aa1b400
[  232.764750] 9fc0: 00000000 bef415f0 c01064ab 00000036 0029bc9c 00be3e30 0023fa68 002403cc
[  232.764758] 9fe0: b6dc708c bef415d4 b6dae950 b6aab8ac
[  232.764766] ---[ end trace 675cb1b5b812c857 ]---
```

It does not work in 4.19-rc8 either, but it behaves differently. This is what I see during boot:
```
[  163.047446] ------------[ cut here ]------------
[... cut ...]
root@raspberrypi:~# vcgencmd measure_clock pixel
frequency(29)=220430000
root@raspberrypi:~# vcgencmd measure_clock hdmi
frequency(9)=299766000
```
With vc4, after applying modeline with `xrandr`:
```
root@raspberrypi:~# vcgencmd measure_clock pixel
frequency(29)=235000
root@raspberrypi:~# vcgencmd measure_clock hdmi
frequency(9)=249513000
```
HDMI clocks are very close to limits that i set in both cases. Pixel clock with framebuffer is equal to my `hdmi_timings` setting, but for vc4 that small value does not make sense to me.