anholt/linux #142: Kernel 4.14.7X, vc4-drm soc:gpu: failed to allocate buffer with size 16777216

rpiri, 2018-10-11:
I spend much time to get raspi working with HW acceleration and chromium-browser in kiosk-mode with no success. I have the HW acceleration (vc4-kms, cma) running without any issues under chrome://gpu. With kiosk-mode I have cma memory allocation errors inside kernel.log. The error comes up after about 5 minutes and result in a freezing screen.

Here is my configuration:
config.txt:
dtoverlay=vc4-kms-v3d,cma-256
avoid_warnings=2
cma_lwm=16
cma_hwm=256
gpu_mem=128

cmdline.txt:
cma=256M@256M

00-rpi-vars:
CHROMIUM_FLAGS="${CHROMIUM_FLAGS} \
--disable-smooth-scrolling \
--disable-quic \
--enable-fast-unload \
--enable-tcp-fast-open \
--enable-offline-auto-reload \
--enable-checker-imaging \
--enable-native-gpu-memory-buffers \
--enable-gpu-rasterization \
--ignore-gpu-blacklist \
--enable-webgl \
"
/home/pi/.config/lxsession/LXDE-pi/autostart:
@chromium-browser --kiosk  --noerrdialogs --incognito --noerrdialogs --disable-infobars --disable-session-crashed-bubble  --ignore-certificate-errors --disable-restore-session-state --no-first-run --fast --fast-start --disable-translate http://mydomain/index.php

kern.log:
…Oct 11 16:00:09 raspi-tv-debug kernel: [  263.537210] vc4-drm soc:gpu: failed to allocate buffer with size 15294464
Oct 11 16:00:09 xxxx kernel: [  263.799190] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
Oct 11 16:00:09 xxxx kernel: [  263.799263] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
Oct 11 16:00:09 xxxx kernel: [  263.799321] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
Oct 11 16:00:09 xxxx kernel: [  263.799371] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
Oct 11 16:00:09 xxxx kernel: [  263.799376] [drm]                         kernel:   6892kb BOs (1)
Oct 11 16:00:09 xxxx kernel: [  263.799379] [drm]                            V3D: 205568kb BOs (257)
Oct 11 16:00:09 xxxx kernel: [  263.799383] [drm]                     V3D shader:    232kb BOs (58)
Oct 11 16:00:09 xxxx kernel: [  263.799386] [drm]                           dumb:     48kb BOs (3)
Oct 11 16:00:09 xxxx kernel: [  263.799392] vc4_v3d 3fc00000.v3d: Failed to allocate memory for tile binning: -12. You may need to enable CMA or give it more memory.

stschake, 2018-10-11:
You have simply run out of graphics memory.

rpiri, 2018-10-11:
I am not sure that you are right. I adjusted the entries in config.txt and cmdline.txt til no errors coming up without running chromium-browser. Then starting chromium in kiosk mode and after about 5 minutes, the errors will appear.

rpiri, 2018-10-11:
Is there a hack for this problem without having to compile a patched kernel by myself?