anholt/linux #143: DSI on cm3 not working

cahz, 2018-10-20:
I am trying to get a DSI display working on the Raspberry Pi Compute Module 3 (cm3). I have added the display to the device tree and enabled vc4 as a dt overlay. However, vc4 is failing on boot. I have tried raspberrypi@b839be4 (rpi-4.14.y) and raspberrypi@c2f4561 (rpi-4.19.y). On 4.14 there is a NULL Pointer occuring in [0], on 4.19 vc4 fails on unbinding the hdmi interface.

On which branch can I expect a working DSI interface for the standard rpi display?

[0] https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/gpu/drm/drm_atomic_helper.c#L156

lategoodbye, 2018-10-22:
Please always report the exact kernel version and full stacktrace in case of a serious error. 

Did you try this [pull request](https://github.com/raspberrypi/linux/pull/2693) and the referenced issue?

cahz, 2018-10-23:
Thanks for pointing out the PR. I looked through some PRs, but must have missed this one. 

Booting up and loading the driver seems better now and there are no errors in the logs. As I don't have the display here at the moment, I cannot confirm functionality yet.