anholt/linux #145: twivc4  RSS feed not accepted by Outlook

mi-hol, 2019-02-20:
I love to read the news updates but with the low frequency it gets increasingly hard to catch them.
This is complicated by the fact that Microsoft Outlook 2016 doesn't accept the RSS feed as valid.
The number of available RSS newsreader is shrinking so anders might have the same issue.
Is this an issue that could be easily fixed to have a convenient approach for your corporate readers to get the news?

[image]

anholt, 2019-02-26:
Sorry, I don't have time to debug RSS from the third-party stack I'm using for the blog.   I just cloned the blog from this week in servo, so you may be able to find fixes by chasing their git history compared to when I forked.

Closing, since this bug is on the wrong repository.

closed 2019-02-26 by anholt

mi-hol, 2019-03-06:
Hi Eric,

sorry to bother you but the VC4 blog title contains the link https://github.com/anholt/linux/wiki/VC4 for reporting bugs. I'm unclear on what repository I should file the bug. 
Please be so kind and provide the link to avoid misunderstandings.

mi-hol, 2019-03-06:
I've found the file feed.xml and filed issue https://github.com/Manishearth/blog.servo.org/issues/1

mi-hol, 2019-03-07:
Just notice that the error was caused by Exchange or Outlook and got fixed in the 1901 release