anholt/linux #146: VC4 mode Xorg.0.log gets spamed

Dark-Sky, 2019-03-17:
Was told to report here by @lategoodnight:

Using VC4 mode my /var/log/Xorg.0.log gets hammered with:

```
[   901.979] (EE) modeset(0): Failed to get GBM bo for flip to new front.
[   901.979] (EE) modeset(0): present flip failed
```
For now I am preventing it with /etc/X11/xorg.conf.d/20-displaylink.conf:

```
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```
I am using the Raspian kernel 5.0.2 but have noticed it on previous Raspian kernels on my rpi 3/3b+ devices both 32bit and aarch64 Arch Linux images. The kernels are compiled here locally from the Raspian github.

anholt, 2019-03-18:
Is a displaylink adapter actually involved somehow?

`echo 0x1f > /sys/module/drm/parameters/debug` and capturing dmesg after the failure may help figure out what's going on.

Dark-Sky, 2019-03-18:
[image]

anholt, 2019-03-19:
You'll want to attach the whole dmesg so you get the piece of it that actually includes the failing call.  Also, still need an answer on why you're talking about displaylink.

Dark-Sky, 2019-03-19:
Other than many many lines like above these creep in every once in a while:

```
[ 1632.198735] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, DRM_IOCTL_CRTC_GET_SEQUENCE
[ 1632.198821] [drm:drm_vblank_enable [drm]] enabling vblank on crtc 2, ret: 0
[ 1632.198919] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, VC4_SUBMIT_CL
[ 1632.199132] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, VC4_WAIT_SEQNO
[ 1632.199657] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, DRM_IOCTL_CRTC_QUEUE_SEQUENCE
[ 1632.206720] [drm:drm_handle_vblank [drm]] vblank event on 97324, current 97324
[ 1632.207134] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, VC4_SUBMIT_CL
[ 1632.207294] [drm:drm_ioctl [drm]] pid=197, dev=0xe200, auth=1, VC4_WAIT_SEQNO
[ 1632.223437] [drm:vblank_disable_fn [drm]] disabling vblank on crtc 2
```
The reason for the /etc/X11/xorg.conf.d/20-displaylink.conf above was a result of searching on the internet that some one posted having the same spamming problem in the Xorg log file. All I know is it stops the spamming in the log.

All I use an HDMI cable connected to my VIZIO TV for a monitor.

The link to the dmesg file below is from boot up and is a total of about 2 minutes after enabling the debugging messages:

https://drive.google.com/file/d/1nGDFzzgmjY2_-dIIhDw3YromLtbd-avM/view?usp=sharing

This is my Xorg log:

https://drive.google.com/file/d/1TRCE1_ewfgK-esqCrzcPAXsf9xZtBxv0/view?usp=sharing

anholt, 2019-03-19:
definitely expected dmesg errors, I was surprised.  Well, I guess at this point I'd need to know how to reproduce the problem -- what specific set of software you're using and what you do with it to see this error.

Dark-Sky, 2019-03-19:
I finally found the problem by reading the thread here:

https://bugzilla.redhat.com/show_bug.cgi?id=1645553

For the sake of brevity It seems to have to do with the mate desktop (which I run on everything here) and the combination of using KMS and having window manager compositor enable. I disabled the window manager compositor and the spamming stopped:

[image]

anholt, 2019-04-04:
OK, sounds like this is https://gitlab.freedesktop.org/xorg/xserver/merge_requests/131, closing.

closed 2019-04-04 by anholt