anholt/linux #149: vc4-fkms-v3d slows down Xorg startup

BigMuscle85, 2019-04-18:
I reported in raspberry/firmware, but this is probably more related place. After enabling KMS fake driver (dtoverlay=vc4-fkms-v3d), the Xorg startup is much slower. 

We are able to boot RPi3B+ (with 7" Pi Touch screen) into our GTK+2 application (from .xinitrc) in 4.5 secs with legacy driver, but after enabling fake KMS driver, the boot takes threetimes more time. The issue is not related to our application. It happens even where starting e.g. lxterminal. I can notice two delays in Xorg.0.log.

One short 800 ms delay directly when glamor driver loads:

> [     8.509] 	ABI class: X.Org ANSI C Emulation, version 0.4
> [     8.509] (II) glamor: OpenGL accelerated X.org driver based.
> [     9.394] (II) glamor: EGL version 1.4 (DRI2):
> [     9.443] (II) modeset(0): glamor initialized

One long delay at the end of the log:

> [     9.992] (II) This device may have been added with another device file.
> [    15.775] (II) modeset(0): Disabling kernel dirty updates, not required.

This second delay is random, sometimes is shorter, sometimes very long. I noticed that application started via .xinitrc is already running at this time ("ps aux" shows it) but its window is not visible on the screen until this "Disabling kernel updates" log message appears.

I also noticed that disabling HDMI output ("tvservice -o") increases this Xorg delay rapidly and it also takes a minutes to device restart (long time of black screen before device actually reboots after "sudo reboot"). No problem with legacy driver.

> [    5.333] (II) glamor: OpenGL accelerated X.org driver based.
> [     6.217] (II) glamor: EGL version 1.4 (DRI2):
> [     6.267] (II) modeset(0): glamor initialized
> [    13.910] (II) modeset(0): Output HDMI-1 has no monitor section
> [    13.910] (II) modeset(0): EDID for output HDMI-1
> [    13.910] (II) modeset(0): Printing probed modes for output HDMI-1
> ...
> [   13.916] (==) Depth 24 pixmap format is 32 bpp
> [    14.165] (==) modeset(0): Backing store enabled
> [    14.165] (==) modeset(0): Silken mouse enabled
> [    14.168] (II) modeset(0): RandR 1.2 enabled, ignore the following RandR disabled message.
> [    45.270] (II) modeset(0): [DRI2] Setup complete
> [    45.270] (II) modeset(0): [DRI2]   DRI driver: vc4
> [    45.270] (II) modeset(0): [DRI2]   VDPAU driver: vc4
> [    45.270] (--) RandR disabled
> [    45.295] (II) SELinux: Disabled on system
> [    45.312] (II) AIGLX: enabled GLX_MESA_copy_sub_buffer
> [    45.312] (II) AIGLX: enabled GLX_ARB_create_context
> [    45.312] (II) AIGLX: enabled GLX_ARB_create_context_profile
> [    45.312] (II) AIGLX: enabled GLX_EXT_create_context_es{,2}_profile
> ...
> [    45.556] (II) config/udev: Adding input device FT5406 memory based driver (/dev/input/mouse0)
> [    45.556] (II) No input driver specified, ignoring this device.
> [    45.556] (II) This device may have been added with another device file.
> [    76.630] (II) SYN_DROPPED event from "FT5406 memory based driver" - some input events have been lost.
> [   125.513] (II) modeset(0): Disabling kernel dirty updates, not required.

Application appears after 125 seconds since boot!

anholt, 2019-04-19:
fkms is a disgusting awful hack in the downstream tree.  Does kms have the same issues?

(probably, I'm just not going to even pretend to support fkms if it's an issue there)

If the CPU is busy during the delays, you should probably try to use sysprof or perf to look into it.

BigMuscle85, 2019-04-24:
I tested VC4-KMS-V3D driver today and it seems to have the same issue. Xorg loads much slower than with generic driver.

> [     6.946] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[     6.993] (II) Module glamoregl: vendor="X.Org Foundation"
[     6.993]    compiled for 1.19.2, module version = 1.0.0
[     6.993]    ABI class: X.Org ANSI C Emulation, version 0.4
[     6.993] (II) glamor: OpenGL accelerated X.org driver based.
[     7.926] (II) glamor: EGL version 1.4 (DRI2):
[     7.978] (II) modeset(0): glamor initialized
[     8.004] (II) modeset(0): Output HDMI-1 has no monitor section
[     8.004] (II) modeset(0): Output Composite-1 has no monitor section
[     8.029] (II) modeset(0): EDID for output HDMI-1
[     8.029] (II) modeset(0): Manufacturer: DEL  Model: f016  Serial#: 826486860
[     8.029] (II) modeset(0): Year: 2010  Week: 31
[     8.029] (II) modeset(0): EDID Version: 1.3
...
[     8.038] (II) UnloadModule: "fbdev"
[     8.038] (II) Unloading fbdev
[     8.039] (II) UnloadSubModule: "fbdevhw"
[     8.039] (II) Unloading fbdevhw
[     8.039] (==) Depth 24 pixmap format is 32 bpp
[     8.292] (==) modeset(0): Backing store enabled
[    10.863] (==) modeset(0): Silken mouse enabled
[    10.886] (II) modeset(0): RandR 1.2 enabled, ignore the following RandR disabled message.
[    12.243] (II) modeset(0): [DRI2] Setup complete
[    12.243] (II) modeset(0): [DRI2]   DRI driver: vc4
[    12.243] (II) modeset(0): [DRI2]   VDPAU driver: vc4
[    12.243] (--) RandR disabled
[    12.267] (II) SELinux: Disabled on system
...
[    12.678] (II) This device may have been added with another device file.
[    13.717] (II) modeset(0): Disabling kernel dirty updates, not required.

But I cannot test how it behaves after disabling HDMI output ("tvservice -o"), because RPi touch display does not work with this driver so HDMI monitor needs to be connected.