anholt/linux #150: vc4 does not use full display

jonas-hagen, 2019-04-24:
I try to use a 1440x2560 display with the Raspberry Pi model 3B+. The display works in portrait mode (`hdmi_timings=1440 1 70 35 45 2560 1 12 2 2 0 0 0 50 0 204792000 3`).

With the legacy driver, the display works as expected at full resolution. When I switch to the vc4-fkms driver, only the middle third of the display is used at an aspect of 4:3. Is it possible that the vc4 driver somehow can not handle portrait mode framebuffers (`framebuffer_width=1440 framebuffer_height=2560`)?

lategoodbye, 2019-04-24:
Could you please at least provide the kernel version / tree you are using?

jonas-hagen, 2019-04-24:
Sorry, completely forgot to mention. I am using the latest Raspberry Pi build
```
uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

I could try to build a kernel and try again. With which kernel version (which branch of this repo) should I try?

anholt, 2019-04-24:
In `drivers/gpu/drm/vc4/vc4_kms.c` you'll find `max_width`/`max_height` settings.  You can try bumping those up to 4095 -- I see enough bits in `SCALER_DISPCTRLX_WIDTH`/`HEIGHT` and `SCALER_POS1_SCL_WIDTH` for that.  >1920 widths may take adjusting the COB allocations between the HVS channels.

jonas-hagen, 2019-04-25:
I tried increasing the `max_` settings just now (see [patch](https://github.com/jonas-hagen/linux/commit/532db9d4dad1a5f4c2e159dd49b6b16ca76cf597) ) and it seems to work! At least I can use the full screen and glxgears runs smoothly. I will now make further tests with full screen opengl.
Thanks!

jonas-hagen, 2019-04-25:
In the virtual terminal (boot screen) the full screen is used from top to bottom, also with `startx` I get a fullscreen display and can move the mouse from top to bottom. But as soon as I try to start `glxgears` (or other openGL applications) in fullscreen mode or appropriate dimensions, the screen stays black or seems to be limited to approx. 1900 pixels in height.
Could you elaborate a bit on what you mean by ">1920 widths may take adjusting the COB allocations between the HVS channels."?

anholt, 2019-04-25:
vc4 has a 3D rendering texture size limit of 2048.  You'll start getting OpenGL errors if you try to exceed that (as X11 might do if we're missing checks)

jonas-hagen, 2019-04-26:
Ok thank you for pointing this out. To my understanding (from reading the code) the view port is limited to 4096 pixels in width and height because of fixed point arithmetic, which is large enough. Texture size limits should not matter when creating a window. Or do I miss something?

Anyways, thanks for your help! I would be motivated to test any ideas you have to overcome this limit, but if its more involved or not a priority, you can close the issue.

anholt, 2019-04-30:
If you're using X11, rendering your windows will involve texturing for various operations, so it is a limit on screen size.

anholt, 2019-05-02:
I'd love to review someone's (tested) patches for >2k display on vc4, regardless of whether you can do GL with it.  I probably won't have time to work on that, though.