anholt/linux #152: Is vc4hdmi ALSA sound card supposed to work by now?

vanfanel, 2019-05-30:
I did a similar report on the raspberry pi kernel repo, but I guess this is a better place since it is @anholt work.

I use the vc4 overlay, and I get two ALSA "sound cards": "ALSA", which is created by the bcm2835 module (dtparam=audio=on), and "vc4hdmi".
The "ALSA" named card works, of course. But the "vc4hdmi" card does not: alsamixer reports it has no controls, and no software can output audio using that card.
I can get ONLY the "vc4hdmi" card if I disable "dtparam=audio=on" or otherwise prevent the bcm2835 module from loading, and then ALSA only reports the "vc4hdmi" card on the system, but even if it is the only card, it does not work either (no controls, libraries fail to open audio device, etc).

So, in a nutshell, does the "vc4hdmi" ALSA card work already?
Thing is, Eric Anholt already reported it is in kernel years ago, but he did not mention that it should be working.

I am using the latest kernel, rpi 4.19.46.

Thanks!

anholt, 2019-05-30:
It should work, but you do need the alsalib config file that is in rpi downstream, and I think may have made it upstream.

vanfanel, 2019-05-30:
@anholt : An alsalib config file is an asoundrc file or an /etc/asound.conf file?